advancedtelematic/meta-updater #42: Plans to add a master branch for bleeding edge development?

gizero, 2017-04-07:
Is there any plan to have a master branch where potentially merging patches for the upcoming pyro release of Yocto? I'm planning to add support for meta-updater in one of my project which is experimental and currently building every layer from master branches. I did no test yet with the morty branch for meta-updater on that setup, but I expect some glitches with recent changes in oe-core master and friends...

OYTIS, 2017-04-09:
Hi @gizero,

we had such plans, good to know that there is a demand for that. I've created a repo manifest (now in a branch, but will be in master soon after a due review process), you can try it with
```
repo init -u https://github.com/advancedtelematic/updater-repo.git -m poky-master.xml -b feat/pokymaster
repo sync
source meta-updater/scripts/envsetup.sh qemux86-64
```
I've tested it on qemux86-64 and it works fine to a first approximation.

OYTIS, 2017-04-11:
OK, so now the manifest is in master, you can init your local copy as:
`repo init -u https://github.com/advancedtelematic/updater-repo.git -m poky-master.xml`

closed 2017-04-11 by OYTIS