advancedtelematic/meta-updater #69: meta-updater/classes/sota_none.bbclass spits out too many warning messages

dl9pf, 2017-05-25:
As this warning message is in the global namespace any task/action will print this out. Over and over.
Also this means we slow-down processing as this is always executed and not just once.

dl9pf, 2017-05-25:
To reproduce just build for an unsupported board.

cajun-rat, 2017-05-26:
Good point. Easiest fix is likely to nuke the warning altogether, but maybe we can persuade it to print only once.

dl9pf, 2017-06-29:
Ping?

cajun-rat, 2017-06-29:
Oops, @OYTIS, Am I right in thinking we can just nuke the whole block?

~~~
python __anonymous() {
	bb.warn("SOTA functionality is not yet supported for your machine")
}
~~~

dl9pf, 2017-06-29:
Well, we should somehow warn the user once, but right now this message is printed not once but for *every* task. Thus like 10000 times.

cajun-rat, 2017-06-30:
Fixed in #93 and #94 , thank you @dl9pf !

closed 2017-06-30 by cajun-rat