advancedtelematic/meta-updater #125: The meta-updater layer brakes the cron daemon

guyshapiro, 2017-08-15:
The 'cronie' recipe creates the directory `/var/spool/cron/` ([by this command](https://github.com/openembedded/openembedded-core/blob/5a25ed1071f0d9b7d95edcc2b5b4545f960d5f95/meta/recipes-extended/cronie/cronie_1.5.1.bb#L60)).
This directory is removed somewhere during the OSTree image creation.
As a result, during boot the crond daemon fails to start with the error:
```
Starting crond: /var/spool/cron: No such file or directory
/var/spool/cron: mkdir: No such file or directory
FAIL
```

Why does it happen?
How can we make the cron daemon work with the meta-updater layer?

Thanks.

cajun-rat, 2017-08-18:
Hi Guy,

The root cause of this is because /var special. It is persisted between updates, and thus isn't updated by OSTree.  Ideally recipes would create any temporary directories they need automatically at run time--this is required if the package isn't in the initial image that is flashed to the device, because we don't know about them when writing the initial image.

There are some special cases handled in [image_types_ota.bbclass](https://github.com/advancedtelematic/meta-updater/blob/morty/classes/image_types_ota.bbclass#L99), which are not really a good solution to the problem, but are required until 'stateless' becomes the norm.

Are you using systemd?  If so, I would propose changing the cronie recipe to use [systemd-tmpfiles](https://www.freedesktop.org/software/systemd/man/tmpfiles.d.html) and create this directory at boot time.

guyshapiro, 2017-08-23:
My usage for the daemon is for recipes that have pre-defined tasks (like [ntp](https://git.congatec.com/yocto/meta-openembedded/blob/morty/meta-networking/recipes-support/ntp/ntp_4.2.8p8.bb#L158)). For that, lets move the directory to a path managed by ostree and not a temporary one.

I'll try to modify the cronie recipe to use `/usr/share/cronie-spool/`.
Additionally, I'll add step on image_types_ostree to move the files from `/var/spool/cron`.

Does this make sense?

guyshapiro, 2017-11-06:
Closed on #132

closed 2017-11-06 by guyshapiro