advancedtelematic/meta-updater #215: u-boot error with qemux86-64 in rocko branch

chbae, 2017-12-18:
FYI @leon-anavi 

I'm trying to upgrade AGL to rocko but I got the below error in qemux86-64.

|    x86_64-agl-linux-ld.bfd   -Bsymbolic -Bsymbolic-functions -m elf_i386 --emit-relocs   -r -o spl/common/built-in.o spl/common/common_fit.o spl/common/miiphyutil.o spl/common/fdt_support.o spl/common/console.o spl/common/dlmalloc.o spl/common/malloc_simple.o spl/common/image.o spl/common/image-fdt.o spl/common/image-fit.o spl/common/memsize.o spl/common/stdio.o spl/common/cli.o spl/common/command.o spl/common/s_record.o spl/common/xyzModem.o
|    rm -f spl/drivers/sysreset/built-in.o; x86_64-agl-linux-ar rcs spl/drivers/sysreset/built-in.o
| make -f /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=spl/drivers/timer
|    x86_64-agl-linux-ld.bfd   -Bsymbolic -Bsymbolic-functions -m elf_i386 --emit-relocs   -r -o spl/drivers/rtc/built-in.o spl/drivers/rtc/rtc-uclass.o spl/drivers/rtc/date.o spl/drivers/rtc/mc146818.o
|   x86_64-agl-linux-gcc  --sysroot=/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/drivers/pci/.pci_x86.o.d  -nostdinc -isystem /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/x86_64-agl-linux/../../lib/x86_64-agl-linux/gcc/x86_64-agl-linux/7.2.0/include -Iinclude  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/arch/x86/include -include /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/spl/drivers/pci -Ispl/drivers/pci -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Werror=date-time -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-toplevel-reorder -fno-dwarf2-cfi-asm -march=i386 -m32 -mregparm=3 -D__I386__ -ffunction-sections -fvisibility=hidden -pipe    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(pci_x86)"  -D"KBUILD_MODNAME=KBUILD_STR(pci_x86)" -c -o spl/drivers/pci/pci_x86.o /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/drivers/pci/pci_x86.c
|   x86_64-agl-linux-gcc  --sysroot=/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/drivers/pci/.pci_auto_common.o.d  -nostdinc -isystem /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/x86_64-agl-linux/../../lib/x86_64-agl-linux/gcc/x86_64-agl-linux/7.2.0/include -Iinclude  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/arch/x86/include -include /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/spl/drivers/pci -Ispl/drivers/pci -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Werror=date-time -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-toplevel-reorder -fno-dwarf2-cfi-asm -march=i386 -m32 -mregparm=3 -D__I386__ -ffunction-sections -fvisibility=hidden -pipe    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(pci_auto_common)"  -D"KBUILD_MODNAME=KBUILD_STR(pci_auto_common)" -c -o spl/drivers/pci/pci_auto_common.o /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/drivers/pci/pci_auto_common.c
|   x86_64-agl-linux-gcc  --sysroot=/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/drivers/timer/.timer-uclass.o.d  -nostdinc -isystem /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/x86_64-agl-linux/../../lib/x86_64-agl-linux/gcc/x86_64-agl-linux/7.2.0/include -Iinclude  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/arch/x86/include -include /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/spl/drivers/timer -Ispl/drivers/timer -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Werror=date-time -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-toplevel-reorder -fno-dwarf2-cfi-asm -march=i386 -m32 -mregparm=3 -D__I386__ -ffunction-sections -fvisibility=hidden -pipe    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(timer_uclass)"  -D"KBUILD_MODNAME=KBUILD_STR(timer_uclass)" -c -o spl/drivers/timer/timer-uclass.o /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/drivers/timer/timer-uclass.c
|   x86_64-agl-linux-gcc  --sysroot=/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/drivers/timer/.tsc_timer.o.d  -nostdinc -isystem /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/x86_64-agl-linux/../../lib/x86_64-agl-linux/gcc/x86_64-agl-linux/7.2.0/include -Iinclude  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/arch/x86/include -include /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/spl/drivers/timer -Ispl/drivers/timer -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Werror=date-time -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-toplevel-reorder -fno-dwarf2-cfi-asm -march=i386 -m32 -mregparm=3 -D__I386__ -ffunction-sections -fvisibility=hidden -pipe    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(tsc_timer)"  -D"KBUILD_MODNAME=KBUILD_STR(tsc_timer)" -c -o spl/drivers/timer/tsc_timer.o /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/drivers/timer/tsc_timer.c
|   x86_64-agl-linux-gcc  --sysroot=/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/drivers/pci/.pci_common.o.d  -nostdinc -isystem /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/x86_64-agl-linux/../../lib/x86_64-agl-linux/gcc/x86_64-agl-linux/7.2.0/include -Iinclude  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/arch/x86/include -include /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git/spl/drivers/pci -Ispl/drivers/pci -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Werror=date-time -ffunction-sections -fdata-sections -fno-strict-aliasing -fomit-frame-pointer -fno-toplevel-reorder -fno-dwarf2-cfi-asm -march=i386 -m32 -mregparm=3 -D__I386__ -ffunction-sections -fvisibility=hidden -pipe    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(pci_ [... cut ...]
| make: Leaving directory '/home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/git'
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ubuntu/work/agl/agl/build/tmp/work/qemux86_64-agl-linux/u-boot/1_2017.09-r0/temp/log.do_compile.11855)
ERROR: Task (/home/ubuntu/work/agl/agl/poky/meta/recipes-bsp/u-boot/u-boot_2017.09.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 471 tasks of which 464 didn't need to be rerun and 1 failed.

leon-anavi, 2017-12-20:
Hi @chbae,

Today I have reproduced the same issue with recipe u-boot_2017.09.bb using Poky release Rocko. I am looking at it and debugging it.

Best regards, Leon

chbae, 2017-12-22:
@leon-anavi 
Great. Thanks a lot.

leon-anavi, 2017-12-22:
Hi @chbae,

I have reproduced the error u-boot_2017.09.bb in Poky. It is generic and it is not caused by meta-updater. One of the known solutions is to [add multilib and to switch to lib32_u-boot](https://lists.yoctoproject.org/pipermail/yocto/2017-October/038334.html). However, I have fixed it by reusing the u-boot-ota recipe from branch Pyro with an additional patch to make it work on Rocko. The following two GitHub PR fix it in Rocko:

* in meta-updater: https://github.com/advancedtelematic/meta-updater/issues/215
* in meta-updater-qemux86-64: https://github.com/advancedtelematic/meta-updater-qemux86-64/pull/13

Yesterday night I did successfully a build from scratch and an end to end update via ATS Garage. I will repeat the test today.

Best regards,
Leon

chbae, 2017-12-24:
@leon-anavi  Thanks :)

patrickvacek, 2017-12-29:
I think you mean https://github.com/advancedtelematic/meta-updater/pull/216 in meta-updater. :)

leon-anavi, 2017-12-29:
Hi @patrickvacek,

Yes, I had in mind #216 :) 

Thanks,
Leon

OYTIS, 2018-01-18:
@chbae 
cc @leon-anavi @patrickvacek 

Sorry, just taken a closer look at the logs and understood that you're using u-boot recipe from AGL, not from meta-updater-qemux86-64, so some important patched do not get applied. Is there an issue in AGL JIRA fro the problem, so that I could refer to it in my PR?
I'm going to revert switch to u-boot-ota shortly, because it doesn't have some other changes that are necessary to work with latest QEMU.

leon-anavi, 2018-02-21:
Hi,

GitHub PR https://github.com/advancedtelematic/meta-updater-qemux86-64/pull/17 offers a solution with a patch that removes the inclusion of the libgcc math functions and replaces them by functions coded in C, taken from the U-Boot (git rev aa7839b39c2ee77f9ab8c393c56b8d812507dbb7) and coreboot projects.

I have tested successfully end to end updates with this GitHub PR using Poky Rocko and AGL master which is now also based on release Rocko of the Yocto Project and OpenEmbedded.

Thanks,
Leon

OYTIS, 2018-04-30:
Can we consider the issue solved?
@chbae @leon-anavi @cajun-rat

leon-anavi, 2018-04-30:
Hi @OYTIS,

Yes, I think so because of the GitHub PR that has been merged but I haven't tested end to end updates in AGL for more than a month.

Best regards,
Leon