advancedtelematic/meta-updater #269: pyro: minnowboard-efi-startup no longer exists ...

dl9pf, 2018-03-05:
... but added by  meta-updater/classes/sota_minnowboard.bbclass:IMAGE_INSTALL_append = " minnowboard-efi-startup"

dl9pf, 2018-03-06:
sri, I meant rocko!

dl9pf, 2018-03-06:
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://gerrit.automotivelinux.org/gerrit/" name="agl" pushurl="ssh://gerrit.automotivelinux.org:29418" review="https://gerrit.automotivelinux.org/gerrit/"/>
  <remote fetch="https://github.com/" name="github"/>
  <remote fetch="git://git.openembedded.org/" name="openembedded"/>
  <remote fetch="git://code.qt.io/" name="qt.io"/>
  <remote fetch="git://git.yoctoproject.org/" name="yocto"/>
  
  <default remote="agl" revision="master" sync-j="4"/>
  
  <project name="01org/meta-security-isafw" path="meta-security-isafw" remote="github" revision="489abdc65cefb566d696c8b218aa0b9b99a350ae" upstream="master"/>
  <project name="AGL/meta-agl" path="meta-agl" revision="e33131d0f4e3f4cbc35f2727802acc486e1b0f95" upstream="master"/>
  <project name="AGL/meta-agl-demo" path="meta-agl-demo" revision="85392a4da4cc5a62660c2d93a1e6a7f09ee37f0a" upstream="master"/>
  <project name="AGL/meta-agl-devel" path="meta-agl-devel" revision="12ec74ae5494ba4b14dad732d00a061bc1ffc409" upstream="master"/>
  <project name="AGL/meta-agl-extra" path="meta-agl-extra" revision="ed8c0a7cf8ad14ca6a6c1dcab4d9f936fcd59bdf" upstream="master"/>
  <project name="AGL/meta-renesas" path="meta-renesas" revision="829c848b8f789671095c757d57a3df5079339e74" upstream="master"/>
  <project name="AGL/meta-renesas-rcar-gen3" path="meta-renesas-rcar-gen3" revision="48efd9e418834569f3d41a31e03dccfa9a6e0f27" upstream="master"/>
  <project name="advancedtelematic/meta-updater" path="meta-updater" remote="github" revision="71863b9c41caeba9bc4d602a55b981743d843a5d" upstream="rocko"/>
  <project name="advancedtelematic/meta-updater-qemux86-64" path="meta-updater-qemux86-64" remote="github" revision="06e89749680012564ad8d96bb46668f1b62d1fd0" upstream="rocko"/>
  <project name="boundarydevices/meta-boundary" path="meta-boundary" remote="github" revision="f96f41b2e5beda2b51acb702d082568898b36a68" upstream="rocko"/>
  <project name="kraj/meta-altera" path="meta-altera" remote="github" revision="565d8443d6dcba4e25924ba049ba81e080f87884" upstream="rocko"/>
  <project name="meta-freescale" remote="yocto" revision="71a54c7b50221800fa4c90c30a425c2f119db3bc" upstream="rocko"/>
  <project name="meta-gplv2" remote="yocto" revision="c7f97f199a6dd54d3f369f0465a6bfd2cfa739b9" upstream="rocko"/>
  <project name="meta-intel" remote="yocto" revision="5415ec29c7745fafb4b7311558d206f9c81fcae4" upstream="rocko"/>
  <project name="meta-oic" remote="yocto" revision="70d5bec59a3b8aabaa4a69f1d8513ff22d86b473" upstream="1.2.1"/>
  <project name="meta-openembedded" remote="openembedded" revision="a65c1acb1822966c3553de9fc98d8bb6be705c4e" upstream="rocko"/>
  <project name="meta-qcom" remote="yocto" revision="d1499bd50a3c91de0bf331009e0beed93739a926" upstream="rocko"/>
  <project name="meta-qt5/meta-qt5" path="meta-qt5" remote="github" revision="db881bb4917628ad3eb1fa7c4f27915cf2ae60f4" upstream="rocko"/>
  <project name="meta-raspberrypi" remote="yocto" revision="510009f0c825cd7992d2d37e9ed0007f6c3be723" upstream="rocko"/>
  <project name="meta-security" remote="yocto" revision="ab8d219c43099bc8d9864a082dad402ecfb7f389" upstream="master"/>
  <project name="meta-ti" remote="yocto" revision="ed83a43c6a76875ee5f0388b3b60a28f2a373a10" upstream="rocko"/>
  <project name="meta-virtualization" remote="yocto" revision="4277759428e96605b8dbe95a43891e217ae8d399" upstream="rocko"/>
  <project name="phongt/meta-sdl" path="meta-sdl" remote="github" revision="60c9fe8a4a9c6ca95f222685f8d6248f16236f2a" upstream="release/4.4.0"/>
  <project name="poky" remote="yocto" revision="c2b641c8a0c4fd71fcb477d788a740c2c26cddce" upstream="rocko"/>
</manifest>

dl9pf, 2018-03-06:
``<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://gerrit.automotivelinux.org/gerrit/" name="agl" pushurl="ssh://gerrit.automotivelinux.org:29418" review="https://gerrit.automotivelinux.org/gerrit/"/>
  <remote fetch="https://github.com/" name="github"/>
  <remote fetch="git://git.openembedded.org/" name="openembedded"/>
  <remote fetch="git://code.qt.io/" name="qt.io"/>
  <remote fetch="git://git.yoctoproject.org/" name="yocto"/>
  
  <default remote="agl" revision="master" sync-j="4"/>
  
  <project name="01org/meta-security-isafw" path="meta-security-isafw" remote="github" revision="489abdc65cefb566d696c8b218aa0b9b99a350ae" upstream="master"/>
  <project name="AGL/meta-agl" path="meta-agl" revision="e33131d0f4e3f4cbc35f2727802acc486e1b0f95" upstream="master"/>
  <project name="AGL/meta-agl-demo" path="meta-agl-demo" revision="85392a4da4cc5a62660c2d93a1e6a7f09ee37f0a" upstream="master"/>
  <project name="AGL/meta-agl-devel" path="meta-agl-devel" revision="12ec74ae5494ba4b14dad732d00a061bc1ffc409" upstream="master"/>
  <project name="AGL/meta-agl-extra" path="meta-agl-extra" revision="ed8c0a7cf8ad14ca6a6c1dcab4d9f936fcd59bdf" upstream="master"/>
  <project name="AGL/meta-renesas" path="meta-renesas" revision="829c848b8f789671095c757d57a3df5079339e74" upstream="master"/>
  <project name="AGL/meta-renesas-rcar-gen3" path="meta-renesas-rcar-gen3" revision="48efd9e418834569f3d41a31e03dccfa9a6e0f27" upstream="master"/>
  <project name="advancedtelematic/meta-updater" path="meta-updater" remote="github" revision="71863b9c41caeba9bc4d602a55b981743d843a5d" upstream="rocko"/>
  <project name="advancedtelematic/meta-updater-qemux86-64" path="meta-updater-qemux86-64" remote="github" revision="06e89749680012564ad8d96bb46668f1b62d1fd0" upstream="rocko"/>
  <project name="boundarydevices/meta-boundary" path="meta-boundary" remote="github" revision="f96f41b2e5beda2b51acb702d082568898b36a68" upstream="rocko"/>
  <project name="kraj/meta-altera" path="meta-altera" remote="github" revision="565d8443d6dcba4e25924ba049ba81e080f87884" upstream="rocko"/>
  <project name="meta-freescale" remote="yocto" revision="71a54c7b50221800fa4c90c30a425c2f119db3bc" upstream="rocko"/>
  <project name="meta-gplv2" remote="yocto" revision="c7f97f199a6dd54d3f369f0465a6bfd2cfa739b9" upstream="rocko"/>
  <project name="meta-intel" remote="yocto" revision="5415ec29c7745fafb4b7311558d206f9c81fcae4" upstream="rocko"/>
  <project name="meta-oic" remote="yocto" revision="70d5bec59a3b8aabaa4a69f1d8513ff22d86b473" upstream="1.2.1"/>
  <project name="meta-openembedded" remote="openembedded" revision="a65c1acb1822966c3553de9fc98d8bb6be705c4e" upstream="rocko"/>
  <project name="meta-qcom" remote="yocto" revision="d1499bd50a3c91de0bf331009e0beed93739a926" upstream="rocko"/>
  <project name="meta-qt5/meta-qt5" path="meta-qt5" remote="github" revision="db881bb4917628ad3eb1fa7c4f27915cf2ae60f4" upstream="rocko"/>
  <project name="meta-raspberrypi" remote="yocto" revision="510009f0c825cd7992d2d37e9ed0007f6c3be723" upstream="rocko"/>
  <project name="meta-security" remote="yocto" revision="ab8d219c43099bc8d9864a082dad402ecfb7f389" upstream="master"/>
  <project name="meta-ti" remote="yocto" revision="ed83a43c6a76875ee5f0388b3b60a28f2a373a10" upstream="rocko"/>
  <project name="meta-virtualization" remote="yocto" revision="4277759428e96605b8dbe95a43891e217ae8d399" upstream="rocko"/>
  <project name="phongt/meta-sdl" path="meta-sdl" remote="github" revision="60c9fe8a4a9c6ca95f222685f8d6248f16236f2a" upstream="release/4.4.0"/>
  <project name="poky" remote="yocto" revision="c2b641c8a0c4fd71fcb477d788a740c2c26cddce" upstream="rocko"/>
</manifest>

dl9pf, 2018-03-06:
`<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://gerrit.automotivelinux.org/gerrit/" name="agl" pushurl="ssh://gerrit.automotivelinux.org:29418" review="https://gerrit.automotivelinux.org/gerrit/"/>
  <remote fetch="https://github.com/" name="github"/>
  <remote fetch="git://git.openembedded.org/" name="openembedded"/>
  <remote fetch="git://code.qt.io/" name="qt.io"/>
  <remote fetch="git://git.yoctoproject.org/" name="yocto"/>
  
  <default remote="agl" revision="master" sync-j="4"/>
  
  <project name="01org/meta-security-isafw" path="meta-security-isafw" remote="github" revision="489abdc65cefb566d696c8b218aa0b9b99a350ae" upstream="master"/>
  <project name="AGL/meta-agl" path="meta-agl" revision="e33131d0f4e3f4cbc35f2727802acc486e1b0f95" upstream="master"/>
  <project name="AGL/meta-agl-demo" path="meta-agl-demo" revision="85392a4da4cc5a62660c2d93a1e6a7f09ee37f0a" upstream="master"/>
  <project name="AGL/meta-agl-devel" path="meta-agl-devel" revision="12ec74ae5494ba4b14dad732d00a061bc1ffc409" upstream="master"/>
  <project name="AGL/meta-agl-extra" path="meta-agl-extra" revision="ed8c0a7cf8ad14ca6a6c1dcab4d9f936fcd59bdf" upstream="master"/>
  <project name="AGL/meta-renesas" path="meta-renesas" revision="829c848b8f789671095c757d57a3df5079339e74" upstream="master"/>
  <project name="AGL/meta-renesas-rcar-gen3" path="meta-renesas-rcar-gen3" revision="48efd9e418834569f3d41a31e03dccfa9a6e0f27" upstream="master"/>
  <project name="advancedtelematic/meta-updater" path="meta-updater" remote="github" revision="71863b9c41caeba9bc4d602a55b981743d843a5d" upstream="rocko"/>
  <project name="advancedtelematic/meta-updater-qemux86-64" path="meta-updater-qemux86-64" remote="github" revision="06e89749680012564ad8d96bb46668f1b62d1fd0" upstream="rocko"/>
  <project name="boundarydevices/meta-boundary" path="meta-boundary" remote="github" revision="f96f41b2e5beda2b51acb702d082568898b36a68" upstream="rocko"/>
  <project name="kraj/meta-altera" path="meta-altera" remote="github" revision="565d8443d6dcba4e25924ba049ba81e080f87884" upstream="rocko"/>
  <project name="meta-freescale" remote="yocto" revision="71a54c7b50221800fa4c90c30a425c2f119db3bc" upstream="rocko"/>
  <project name="meta-gplv2" remote="yocto" revision="c7f97f199a6dd54d3f369f0465a6bfd2cfa739b9" upstream="rocko"/>
  <project name="meta-intel" remote="yocto" revision="5415ec29c7745fafb4b7311558d206f9c81fcae4" upstream="rocko"/>
  <project name="meta-oic" remote="yocto" revision="70d5bec59a3b8aabaa4a69f1d8513ff22d86b473" upstream="1.2.1"/>
  <project name="meta-openembedded" remote="openembedded" revision="a65c1acb1822966c3553de9fc98d8bb6be705c4e" upstream="rocko"/>
  <project name="meta-qcom" remote="yocto" revision="d1499bd50a3c91de0bf331009e0beed93739a926" upstream="rocko"/>
  <project name="meta-qt5/meta-qt5" path="meta-qt5" remote="github" revision="db881bb4917628ad3eb1fa7c4f27915cf2ae60f4" upstream="rocko"/>
  <project name="meta-raspberrypi" remote="yocto" revision="510009f0c825cd7992d2d37e9ed0007f6c3be723" upstream="rocko"/>
  <project name="meta-security" remote="yocto" revision="ab8d219c43099bc8d9864a082dad402ecfb7f389" upstream="master"/>
  <project name="meta-ti" remote="yocto" revision="ed83a43c6a76875ee5f0388b3b60a28f2a373a10" upstream="rocko"/>
  <project name="meta-virtualization" remote="yocto" revision="4277759428e96605b8dbe95a43891e217ae8d399" upstream="rocko"/>
  <project name="phongt/meta-sdl" path="meta-sdl" remote="github" revision="60c9fe8a4a9c6ca95f222685f8d6248f16236f2a" upstream="release/4.4.0"/>
  <project name="poky" remote="yocto" revision="c2b641c8a0c4fd71fcb477d788a740c2c26cddce" upstream="rocko"/>
</manifest>
`

dl9pf, 2018-03-06:
```
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://gerrit.automotivelinux.org/gerrit/" name="agl" pushurl="ssh://gerrit.automotivelinux.org:29418" review="https://gerrit.automotivelinux.org/gerrit/"/>
  <remote fetch="https://github.com/" name="github"/>
  <remote fetch="git://git.openembedded.org/" name="openembedded"/>
  <remote fetch="git://code.qt.io/" name="qt.io"/>
  <remote fetch="git://git.yoctoproject.org/" name="yocto"/>
  
  <default remote="agl" revision="master" sync-j="4"/>
  
  <project name="01org/meta-security-isafw" path="meta-security-isafw" remote="github" revision="489abdc65cefb566d696c8b218aa0b9b99a350ae" upstream="master"/>
  <project name="AGL/meta-agl" path="meta-agl" revision="e33131d0f4e3f4cbc35f2727802acc486e1b0f95" upstream="master"/>
  <project name="AGL/meta-agl-demo" path="meta-agl-demo" revision="85392a4da4cc5a62660c2d93a1e6a7f09ee37f0a" upstream="master"/>
  <project name="AGL/meta-agl-devel" path="meta-agl-devel" revision="12ec74ae5494ba4b14dad732d00a061bc1ffc409" upstream="master"/>
  <project name="AGL/meta-agl-extra" path="meta-agl-extra" revision="ed8c0a7cf8ad14ca6a6c1dcab4d9f936fcd59bdf" upstream="master"/>
  <project name="AGL/meta-renesas" path="meta-renesas" revision="829c848b8f789671095c757d57a3df5079339e74" upstream="master"/>
  <project name="AGL/meta-renesas-rcar-gen3" path="meta-renesas-rcar-gen3" revision="48efd9e418834569f3d41a31e03dccfa9a6e0f27" upstream="master"/>
  <project name="advancedtelematic/meta-updater" path="meta-updater" remote="github" revision="71863b9c41caeba9bc4d602a55b981743d843a5d" upstream="rocko"/>
  <project name="advancedtelematic/meta-updater-qemux86-64" path="meta-updater-qemux86-64" remote="github" revision="06e89749680012564ad8d96bb46668f1b62d1fd0" upstream="rocko"/>
  <project name="boundarydevices/meta-boundary" path="meta-boundary" remote="github" revision="f96f41b2e5beda2b51acb702d082568898b36a68" upstream="rocko"/>
  <project name="kraj/meta-altera" path="meta-altera" remote="github" revision="565d8443d6dcba4e25924ba049ba81e080f87884" upstream="rocko"/>
  <project name="meta-freescale" remote="yocto" revision="71a54c7b50221800fa4c90c30a425c2f119db3bc" upstream="rocko"/>
  <project name="meta-gplv2" remote="yocto" revision="c7f97f199a6dd54d3f369f0465a6bfd2cfa739b9" upstream="rocko"/>
  <project name="meta-intel" remote="yocto" revision="5415ec29c7745fafb4b7311558d206f9c81fcae4" upstream="rocko"/>
  <project name="meta-oic" remote="yocto" revision="70d5bec59a3b8aabaa4a69f1d8513ff22d86b473" upstream="1.2.1"/>
  <project name="meta-openembedded" remote="openembedded" revision="a65c1acb1822966c3553de9fc98d8bb6be705c4e" upstream="rocko"/>
  <project name="meta-qcom" remote="yocto" revision="d1499bd50a3c91de0bf331009e0beed93739a926" upstream="rocko"/>
  <project name="meta-qt5/meta-qt5" path="meta-qt5" remote="github" revision="db881bb4917628ad3eb1fa7c4f27915cf2ae60f4" upstream="rocko"/>
  <project name="meta-raspberrypi" remote="yocto" revision="510009f0c825cd7992d2d37e9ed0007f6c3be723" upstream="rocko"/>
  <project name="meta-security" remote="yocto" revision="ab8d219c43099bc8d9864a082dad402ecfb7f389" upstream="master"/>
  <project name="meta-ti" remote="yocto" revision="ed83a43c6a76875ee5f0388b3b60a28f2a373a10" upstream="rocko"/>
  <project name="meta-virtualization" remote="yocto" revision="4277759428e96605b8dbe95a43891e217ae8d399" upstream="rocko"/>
  <project name="phongt/meta-sdl" path="meta-sdl" remote="github" revision="60c9fe8a4a9c6ca95f222685f8d6248f16236f2a" upstream="release/4.4.0"/>
  <project name="poky" remote="yocto" revision="c2b641c8a0c4fd71fcb477d788a740c2c26cddce" upstream="rocko"/>
</manifest>
```

OYTIS, 2018-04-30:
@dl9pf 
Sorry for a really late response. It seems to me that minnowboard-efi-startup was front-ported from pyro on 29th of January and should be in meta-updater-minnowboard now. What we probably forgot to do is to add it to AGL. Is it breaking the builds?
Without this recipe EFI will not find the bootloader on an SD card. I'm not sure how this problem is solved in AGL now.

OYTIS, 2018-05-02:
Looks like agl uses bbappend to the kernel recipe to deploy startup.nsh. I'll take a look at what can be done in meta-updater.

patrickvacek, 2018-11-29:
Is this still a problem? I'm not sure anyone has tried booting a minnowboard in a while...