advancedtelematic/meta-updater #301: image built for RPI3 does not boot - freezes at "Starting the kernel..."

mihaelaapetroaie, 2018-04-20:
Hello, 

I am trying to build meta-updater for RPI3, rocko version, but the image does not boot.
My build configuration:

```
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.2"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "rocko:1648bcafa3d0e46acee61a34d0a07fabb85b1f8d"
meta-virtualization  = "rocko:bd77388f31929f38e7d4cc9c711f0f83f563007e"
meta-oe              = "rocko:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-remote-management = "rocko:1648bcafa3d0e46acee61a34d0a07fabb85b1f8d"
meta-networking      
meta-python          
meta-filesystems     = "rocko:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-raspberrypi     = "rocko:b58ccfc696e4fa6b680ae5ca79960312676d7af9"
meta-rust            = "rocko:a4797129e2ab7f11671f817653b11ed876c2b43c"
oe-meta-go           = "master:514b2a80a2a4235687e92fb28328bb3e7c2d6c74"
meta-java            = "rocko:fbe0b0eb30d93f5fa2101fb015f20c1dc118b4a6"

```

my local.conf file settings:

 ```
MACHINE ??= "raspberrypi3"

 DISTRO_FEATURES_append = " virtualization systemd pam wifi"
 INHERIT += " sota"
 # Prelinking increases the size of downloads and causes build errors
 USER_CLASSES_remove = "image-prelink"
VIRTUAL-RUNTIME_init_manager = "systemd"
 VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
 DISTRO ?= "poky-sota-systemd"
PACKAGE_CLASSES ?= "package_ipk"
IMAGE_FEATURES+= "package-management"

BB_DISKMON_DIRS = "\
 STOPTASKS,${TMPDIR},1G,100K \
STOPTASKS,${DL_DIR},1G,100K \
STOPTASKS,${SSTATE_DIR},1G,100K \
STOPTASKS,/tmp,100M,100K \
ABORT,${TMPDIR},100M,1K \
ABORT,${DL_DIR},100M,1K \
ABORT,${SSTATE_DIR},100M,1K \
 ABORT,/tmp,10M,1K"

> PACKAGECONFIG_append_pn-qemu-native = " sdl"
> PACKAGECONFIG_append_pn-nativesdk-qemu`= " sdl"
> IMAGE_FSTYPE_append="sdimg"

 # Install packages with
 #IMAGE_INSTALL_append = " "
 SERIAL_CONSOLE=""
 ENABLE_SPI_BUS="1"
 ENABLE_I2C="1"
 ENABLE_SPI="1"
 #SKIP_META_VIRT_SANITY_CHECK=1
 PACKAGECONFIG_append_pn-connman = " openvpn vpnc l2tp pptp"
RPI_USE_U_BOOT = "1"
```

my bblayers.conf

 ```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
 # changes incompatibly
 POKY_BBLAYERS_CONF_VERSION = "2" 
 BBPATH = "${TOPDIR}"
 BBFILES ?= ""
 
 BBLAYERS ?= " \
   /rocko_poky/poky/meta \
  /rocko_poky/poky/meta-poky \
   /rocko_poky/poky/meta-virtualization \
  /rocko_poky/poky/meta-openembedded/meta-oe \
   /rocko_poky/poky/meta-remote-management \
   /rocko_poky/poky/meta-openembedded/meta-networking \
   rocko_poky/poky/meta-openembedded/meta-python \
  /rocko_poky/poky/meta-openembedded/meta-filesystems \
   /rocko_poky/poky/meta-raspberrypi \
   /rocko_poky/poky/meta-rust \
   /rocko_poky/poky/oe-meta-go \
  /rocko_poky/poky/meta-java \
   "

```
The sdimg file that was created in morty disappeared, and it was replaced with a wic file that does not boot after I dd it on my sd card. If I remove the meta-updater from my bblayers and re-build the problem seems to disappear.

I also tried to do it from scratch following the tutorial at

>  https://docs.atsgarage.com/quickstarts/raspberry-pi.html#2-create-your-yocto-build-environment

but I am hitting the exact same problem.

I am wondering if anyone hit the same problem and if you have an idea about what the problem could be.

Thank you,
Mihaela

patrickvacek, 2018-04-23:
Hello @mihaelaapetroaie. Are you seeing any warnings when bitbaking the image? And do you get any output from the rpi on the serial console when you try to boot it?

It is unusual to me that your `BBLAYERS` do not include `poky/meta-yocto-bsp`, which is usually a default layer, although I don't know how necessary it really is.

I also do not believe you need to set `RPI_USE_U_BOOT = "1"` in your `local.conf`. We set that automatically in `meta-updater/classes/sota_raspberrypi.bbclass`.

mihaelaapetroaie, 2018-04-23:
Hi, 

Thanks for your reply. I get the same result with and without the meta-yocto-bsp layer, same with the RPI_USE_U_BOOT configuration.

I am getting a warning message:
```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky-sota"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-yocto-bsp       
meta-poky            = "rocko:1648bcafa3d0e46acee61a34d0a07fabb85b1f8d"
meta-virtualization  = "rocko:bd77388f31929f38e7d4cc9c711f0f83f563007e"
meta-oe              = "rocko:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-remote-management = "rocko:1648bcafa3d0e46acee61a34d0a07fabb85b1f8d"
meta-networking      
meta-python          
meta-filesystems     = "rocko:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-raspberrypi     = "rocko:b58ccfc696e4fa6b680ae5ca79960312676d7af9"
meta-updater         = "rocko:156074d9533acf88f102a07701591a4532d007db"
meta-updater-raspberrypi = "rocko:4acd3fac3e13616564c4ea19e09265b30a63dfd6"
meta-rust            = "rocko:a4797129e2ab7f11671f817653b11ed876c2b43c"
oe-meta-go           = "master:514b2a80a2a4235687e92fb28328bb3e7c2d6c74"
meta-java            = "rocko:fbe0b0eb30d93f5fa2101fb015f20c1dc118b4a6"

Initialising tasks: 100% |#################################################################################################| Time: 0:00:08
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: initramfs-ostree-image-1.0-r0 do_image: Android repo tool failed to run; manifest not copied
NOTE: Tasks Summary: Attempted 5101 tasks of which 5048 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.

```

The the serial console the the boot is hanging in:
`Started Update UTMP about System Runlevel changes`

OYTIS, 2018-04-27:
Hello Mihaela,

I've tried building with different configurations and it looks to me that `Started Update UTMP about System Runlevel changes` is the result of setting `SERIAL_CONSOLE` to an empty string, although I can't tell you exactly why systemd and login/bash are using different console settings yet. Is it critical?
For the lack of console output to HDMI, it is a known issue we've probably ignored for too long, we'll try to fix it.

Thanks,
Anton

OYTIS, 2018-05-04:
@mihaelaapetroaie 
The change to meta-updater-raspberrypi has been just merged. If you synchronize your layers and rebuild, the console should be fixed.

closed 2018-05-04 by OYTIS