advancedtelematic/meta-updater #306: do_image_garagesign fails with java 10

Bubu, 2018-05-02:
I get the following error:

```
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_garagesign
| 
| The java installation you have is not up to date
| requires at least version 1.6+, you have
| version 10.0.1
| 
| Please go to http://www.java.com/getjava/ and download
| a valid Java Runtime and install before running .
| 
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_image_garagesign 
```

cajun-rat, 2018-05-03:
Hi,

meta-updater will pick up the version of java from the host environment, so the easiest way to change it is with [update-alternatives](https://askubuntu.com/questions/315646/update-java-alternatives-vs-update-alternatives-config-java#388693)

We've been able to build successfully on Ubuntu 18.04, so this should work.  One thing to be careful of if you have upgraded from an older installation is that the tmp directories and sstate cache will need to be blown away because the host compiler has changed.

Bubu, 2018-05-03:
Thanks, will try this later.

The thing is that it **did** work with java 10 after I edited out the broken version check from the garage-sign startup script :woman_shrugging:

cajun-rat, 2018-05-29:
Internal Jira reference is PRO-5445

patrickvacek, 2018-05-29:
@Bubu Our CI server has the fix now, but until it gets deployed to the production server, you can put something like this into your `local.conf` to force using the fixed version:

`GARAGE_SIGN_VERSION = "cli-0.4.0-26-gbd9fd3c.tgz"`

closed 2018-08-08 by patrickvacek