advancedtelematic/meta-updater #320: [RFC] support upgrading devicetree

gizero, 2018-05-23:
Hi,

I recently had a small patch merged upstream to the ostree repo [1]. This is intended to be the foundation for supporting the upgrade of a system's devicetree alongside the kernel and the ramdisk as for now.

My primary goal is to support at least the simplest scenario where a single devicetree is delivered as part of an ostree deployment and to make the u-boot based boot process aware of the devicetree location to use it at boot. 

We already have some patchsets, still mostly proofs of concept. We successfully used them to test the full flow of delivering a devicetree change through the usual ostree upgrade chain. 

Before moving on to refining and hopefully submitting such patches, it would be nice to have some feedback from you. I couldn't find any evidence of previous efforts geared at a similar goal. Did I miss any? What's your opinion on adding devicetree(s) to ostree-upgradable artifacts at all?

In my initial analysis I noticed what's probably a gray area in the Yocto/OE ecosystem w.r.t. to selecting which devicetree should be used in a system at boot. This is probably related to the lack of a common approach for generating u-boot environments to deliver onto the target. We do have `KERNEL_DEVICETREE` which in my understanding is intended to host all the supported/available devicetree for a given machine, but I'm not aware of a convenient way for selecting which one should be used when the bootable image we build/deliver. This is my two cents. What're the community thoughts on this? 

[1] https://github.com/ostreedev/ostree/pull/1590

patrickvacek, 2018-05-24:
Hello, @gizero. I'm not aware of any work in meta-updater to support this so far, but it seems like a reasonable thing to be able to do. Perhaps @cajun-rat, @OYTIS, or @lbonn have some further insight?

cajun-rat, 2018-05-29:
@gizero Our approach so far for this has been to use FIT images to bundle the device tree and kernel image together. We do this on R-Pi here: https://github.com/advancedtelematic/meta-updater-raspberrypi/commit/94b315d8f69db0c10aee1fcb8626480c4d8626bd

Was there a reason to prefer a separate device tree to FIT images?

fernandoendo, 2018-10-17:
Releasing the DTB in /boot/ostree/poky-*/ would allow a more modular approach than using FIT images. Supporting both approaches would be nice from my point of view.

gizero, 2018-10-19:
@fernandoendo Agreed! I have a working (although quite hackish) solution that has been working for months: I need to find the time to refine it at a minimum to ask someone for review. Would you be interested? Do you already have your own approach? @cajun-rat One of the reasons why devicetrees were originally introduced was to allow updating/changing platform configuration data without the need to deploy a full kernel/modules set every time. I agree this is not happening very frequently, but for sure is the reason why I never took the time to look into FIT, which IMHO looked like a less flexible approach to delivering kernel/DTB bundles... But I might be wrong: I'll take some time to look into FIT again.

patrickvacek, 2020-07-30:
Closing due to lack of feedback.

closed 2020-07-30 by patrickvacek