advancedtelematic/meta-updater #323: build failure on rocko branch

mirzak, 2018-05-28:
Tested a build using:

```
repo init -u https://github.com/advancedtelematic/updater-repo.git -m rocko.xml
```

But it fails on building `aktualizr-native` in the `do_compile` step with errors like:

```
aktualizr-native/1.0+gitAUTOINC+3b89858cf8-7/git/src/sota_tools/ostree_http_repo.cc:39:104: error: use of deleted function ‘OSTreeRef::OSTreeRef(OSTreeRef&&)’
 OSTreeRef OSTreeHttpRepo::GetRef(const std::string &refname) const { return OSTreeRef(*server_, refname); }
```

and

```
aktualizr-native/1.0+gitAUTOINC+3b89858cf8-7/git/src/sota_tools/ostree_ref.h:16:3: error: use of deleted function ‘std::basic_stringstream<char>::basic_stringstream(const std::basic_stringstream<char>&)’
```

There is a lot of them. Can provide full log if necessary.

My environment:

```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.9"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky-sota"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:7e7ee662f5dea4d090293045f7498093322802cc"
meta-updater         = "HEAD:4bcd0509c780a2e244aab4713926a548c176fb5f"
meta-filesystems     
meta-oe              = "HEAD:dacfa2b1920e285531bec55cd2f08743390aaf57"
meta-updater-qemux86-64 = "HEAD:697632ddd98ed7ae3dbd0bd84abb04079767bc56"
```

Debian 8 which is part of the "supported" distro versions for rocko

```
$ cat /etc/issue
Debian GNU/Linux 8 \n \l
```

cajun-rat, 2018-05-28:
It looks like the problem here is that the version of gcc that ships with Jessie doesn't fully support C++11 (or at least has some bugs in the support). Compilation for the target should be fine, because Yocto builds a cross compiler with a recent gcc, but the aktualizr-native recipe will rely on the host gcc.

Is is possible to upgrade gcc on the host to something more recent?  The oldest compiler we test against the default gcc in Debian Stable, so that would be a safe bet.

mirzak, 2018-05-28:
I can confirm that compilation for target works just fine.

FYI my host GCC version:

```
$ gcc --version
gcc (Debian 4.9.2-10+deb8u1) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

But will try with latest debian stable. Though it is a bit unfortunate to deviate from the "Yocto" supported distributions where both Debian 8 and Debian 9 is listed as supported for 2.4 release (rocko). This is now not true when meta-updater is included and raises issues like these.

mirzak, 2018-05-28:
I did also try the `pyro.xml` on the same machine which compiled just fine.

patrickvacek, 2018-11-29:
@mirzak Is this still a problem for you, and if so, do you think the work needed to solve it is substantial?

mirzak, 2018-11-29:
I am not actively working on this any more, but I worked around this with running the latest Debian (Debian 9) instead of Jessie.

patrickvacek, 2018-11-29:
Thanks for following up! Closing for now, then. We can reopen if someone runs into this again.

closed 2018-11-29 by patrickvacek