advancedtelematic/meta-updater #334: btrfs rootfs

mihaelaapetroaie, 2018-06-11:
Hello, 

Is there any easy way to get the rootfs (otaimg) as btrfs instead of ext4?

Thank you!

lbonn, 2018-06-12:
The error you're seeing comes from the usage of `mkfs.btrfs` which differs from `mkfs.ext4` for some reason. It should be `mkfs.btrfs ${DEPLOY_DIR_IMAGE}/${IMAGE_NAME}.otaimg -L otaroot -r ${PHYS_SYSROOT}` (`-r` instead of `-d`)

However, after fixing that, another cryptic error appeared:

```
| ERROR: unable to traverse directory ...build-qemu/tmp/work/qemux86_64-poky-linux/core-image-minimal/1.0-r0/ota-sysroot-VDmyv: 1
| ERROR: error wihle filling filesystem: 1
| ERROR: attempt to start transaction over already running one
| extent buffer leak: start 4440064 len 16384
| extent_io.c:607: free_extent_buffer_internal: BUG_ON `eb->flags & EXTENT_DIRTY` triggered, value 1
```

I don't really where this one could come from.

Jmennius, 2018-06-12:
Hi @mihaelaapetroaie , your issue is probably related to the fact that mkfs.btrfs has slightly different interface than mkfs.ext*. This way it should work:
`mkfs.btrfs --mixed -L otaroot -r ${PHYS_SYSROOT} ${DEPLOY_DIR_IMAGE}/${IMAGE_NAME}.otaimg`

There is a high chance that you will encounter another issue related to mkfs.btrfs which was recently fixed ([in devel branch](https://github.com/kdave/btrfs-progs/commit/01f2380ff23585b1c07d7d53024f0303ed1e85ca)).
If you are going to use u-boot (to load stuff from btrfs) - you will likely run into more issues (see [1](https://patchwork.ozlabs.org/patch/927619/) and [2](https://patchwork.ozlabs.org/patch/925389/)).

mihaelaapetroaie, 2018-06-13:
Hello,

Thank you for your answers. I think I'll keep using ext4 for now. I'll experiment with btrf later, when I'll have a bit more free time.

patrickvacek, 2019-01-16:
Closing due to inactivity and the fact that image generation has been reworked in thud/master, which may make this easier anyway.

closed 2019-01-16 by patrickvacek