advancedtelematic/meta-updater #342: Support tarballs OTA images

falstaff84, 2018-06-15:
To integrate meta-updater with a custom installation systems we need a tarball of the updateable ostree tree. Currently  image_types_ota creates ext4 images only. The class should be extended to support tar. Maybe the generation and naming of the image formats should be hierarchical, e.g. otaimg.ext4/otaimg.tar...

cajun-rat, 2018-06-15:
It should be possible to do that, maybe as some kind of optional step, because we don't want to generate multiple images for everyone.  A hack in image_types_ota that just checked a flag and created the image if necessary would be simple, but perhaps not ideal.

falstaff84, 2018-06-15:
I would suggest to just create a new image type (otaimg.tar) and then move most of the functionality currently in IMAGE_CMD_otaimg into a common function. Then IMAGE_CMD_otaimg ends up calling this function and creating the ext4 fs, and IMAGE_CMD_otaimg.tar the same but creating the tarball in the end...

patrickvacek, 2018-10-09:
@agners Can I close this since the PR was merged and it looks like your needs have been met? Or is there something more related to this that I've overlooked?

agners, 2018-10-09:
Our needs have been met with that PR merged! You can close this issue. Thanks!

closed 2018-10-09 by patrickvacek