advancedtelematic/meta-updater #356: Usage of custom variables for initramfs and bootloader selection

Jmennius, 2018-07-04:
This is regarding the existence of `OSTREE_INITRAMFS_IMAGE`/`TYPES`, `OSTREE_BOOTLOADER` and `OSTREE_KERNEL`. Some are just inconvenient and some prevent you from using features.

For example, `OSTREE_INITRAMFS_IMAGE` effectively builds initramfs image w/o OE actually knowing about it, which makes it impossible to include the initramfs in FIT. And the inclusion of initramfs in FIT is crucial if you want to have trusted/secure boot (e.g. verify a signature of initramfs).

I am willing to improve/fix this and I would like to understand if there is any backstory behind these variables and if you would accept such a change (considering users of meta-updater will need to go through a fairly simple migration process).

OYTIS, 2018-07-09:
Hi @Jmennius,
it would be great if you could improve meta-updater codebase with these changes. The only reason to keep `OSTREE_INITRAMFS_IMAGE` I can imagine is that it prevents user from unintentionally breaking OSTree boot by setting `INITRAMFS_IMAGE` to something else. But this can probably be avoided in a cleaner way.

Jmennius, 2018-07-18:
Hi @OYTIS, see first commit in #360 which solves an issue with `OSTREE_INITRAMFS_IMAGE` and then employs this change to put initramfs in FIT.

agners, 2018-10-18:
@Jmennius with the merge of #360, is this resolved?

Jmennius, 2018-10-19:
@agners, well.. `OSTREE_BOOTLOADER` and `OSTREE_KERNEL` are still there so I would keep this issue as a reminder that this should be refactored.

OYTIS, 2019-01-29:
@Jmennius 

The problem is there are many quite different scenarios of using Yocto/OE, and we try to be able to adapt to all of them.

For instance it's possible to use a customized version of u-boot instead of the one provided by Yocto. `PREFERRED_PROVIDER_virtual/bootloader` will then be set to the name of this customized recipe. But OSTree logic still needs to know that this bootloader is a flavour of u-boot, because everything that matters for OSTree is the difference between U-boot, Grub and Syslinux. That's what `OSTREE_BOOTLOADER` is used for.

As for `OSTREE_KERNEL`, Yocto/OE has no notion of 'the kernel image that will be deployed on the device'. `KERNEL_IMAGENAME` is a good heuristic for it, but there are some corner cases where it doesn't work. IIRC, this variable was inspired by how r[ecipes for Renesas Porter board](https://github.com/CogentEmbedded/meta-rcar/blob/v2.12.0/meta-rcar-gen2/recipes-kernel/linux/linux-dtb-append.inc) generated kernel images packed with dtb files.

While Porter board is quite out of date, we can encounter similar issues in future, and as was said above, Yocto/OE doesn't have a variably that would mean exactly what `OSTREE_KERNEL` means.

patrickvacek, 2020-07-30:
This has been improved in the meantime, and can be reopened if there are still problems.

closed 2020-07-30 by patrickvacek