advancedtelematic/meta-updater #375: lshw: Bus error (core dumped)

maximeborges, 2018-08-30:
Hello,
when starting `aktualizr` i'm getting the following error:
```
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: Aktualizr version 1.0+gitAUTOINC+a94a15e31f starting
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: Reading config: "/usr/lib/sota/conf.d/20-sota_autoprov.toml"
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: Use existing SQL storage: "/var/sota/sql.db"
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: got SendDeviceData command
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: Bus error (core dumped)
Aug 30 09:28:58 imx6ul-opos6uldev aktualizr[424]: Could not execute lshw (is it installed?).
```

Starting `lshw` manually, I'm getting:
```
root@imx6ul-opos6uldev:/sysroot/home/root# lshw
[ 1136.600686] Unhandled fault: external abort on non-linefetch (0x008) at 0xb6f93000
[ 1136.608536] pgd = 01f78776
[ 1136.611367] [b6f93000] *pgd=8d059835, *pte=000e0783, *ppte=000e0e33
Bus error (core dumped)
```

When starting `lshw -disable dmi` it's working fine.

It looks like there is a problem on the `B.02.16`: https://bugs.launchpad.net/ubuntu/+source/lshw/+bug/653082

I updated the `lshw` version in the recipe to `B.02.17` and it's I don't get any error anymore.

Distro: `sumo`

patrickvacek, 2018-08-31:
@maximeborges I haven't seen that problem myself yet, but upgrading seems like a reasonable thing to do. Do you want to make a PR with your changes? That should be easy to test and merge from our side. Thanks for the report!

patrickvacek, 2018-08-31:
(Also: I assume this would affect master and rocko as well, but if you make a PR for any one of those branches, we can port it to all three.)