advancedtelematic/meta-updater #402: Missing DISTRO_FEATURES

fernandoendo, 2018-10-09:
Hello,

I'm building SOTA in a custom distro. I found out that the following line was needed in my configuration:

`DISTRO_FEATURES_append = " systemd"`

Maybe it should come in  conf/distro/sota.conf.inc?

Regards,
Fernando Endo

patrickvacek, 2018-10-09:
> systemd is a build dependency of Aktualizr

That is true. Our usual case is systemd. However, it should be possible to use aktualizr (and ostree) without systemd with some modifications. Neither one is fundamentally dependent on systemd as far as I am aware, but we only have experience using them with systemd. You are welcome to try it, and if you think you have a patch that will enable aktualizr to work flexibly with or without systemd, we would be curious to see it.

I do know that building aktualizr with `-DBUILD_OSTREE=OFF` is easy (and the default on non-Linux systems). I personally know less about what edits would be necessary to the aktualizr and ostree recipes in meta-updater.

cajun-rat, 2018-10-09:
There is an option to build Aktualizr without systemd as a dependency.  There is a cmake option `-DBUILD_SYSTEMD=OFF`, but it isn't currently wired into the recipes in meta-updater.  Adding that to [`EXTRA_OECMAKE`](https://github.com/advancedtelematic/meta-updater/blob/master/recipes-sota/aktualizr/aktualizr_git.bb#L49) might make it build.  Wiring that into the Yocto `DISTRO_FEATURES` shouldn't be _too_ hard.

patrickvacek, 2018-11-29:
@fernandoendo Have you had any luck making this work, or is it still a problem for you?

liuming50, 2019-01-30:
Hi, I am working on enabling PACKAGECONFIG in aktualizr, will send a patch soon, the complete PACKAGECONFIG would be like:
```
PACKAGECONFIG[warning-as-error] = "-DWARNING_AS_ERROR=ON,-DWARNING_AS_ERROR=OFF,"
PACKAGECONFIG[pedantic-warnings] = "-DPEDANTIC_WARNINGS=ON,-DPEDANTIC_WARNINGS=OFF,"
PACKAGECONFIG[code-coverage] = "-DBUILD_WITH_CODE_COVERAGE=ON,-DBUILD_WITH_CODE_COVERAGE=OFF,"
PACKAGECONFIG[ostree] = "-DBUILD_OSTREE=ON,-DBUILD_OSTREE=OFF,ostree,"
PACKAGECONFIG[deb] = "-DBUILD_DEB=ON,-DBUILD_DEB=OFF,dpkg,"
PACKAGECONFIG[p11] = "-DBUILD_P11=ON,-DBUILD_P11=OFF,libp11,"
PACKAGECONFIG[sota-tools] = "-DBUILD_SOTA_TOOLS=ON,-DBUILD_SOTA_TOOLS=OFF,glib-2.0,"
PACKAGECONFIG[opcua] = "-DBUILD_OPCUA=ON,-DBUILD_OPCUA=OFF,boost,"
PACKAGECONFIG[isotp] = "-DBUILD_ISOTP=ON,-DBUILD_ISOTP=OFF,"
PACKAGECONFIG[systemd] = "-DBUILD_SYSTEMD=ON,-DBUILD_SYSTEMD=OFF,systemd,"
PACKAGECONFIG[load-tests] = "-DBUILD_LOAD_TESTS=ON,-DBUILD_LOAD_TESTS=OFF,"
PACKAGECONFIG[fault-injection] = "-DFAULT_INJECTION=ON,-DFAULT_INJECTION=OFF,libfiu"
```
with the above setting, we can also clean up the DEPENDS/RDEPENDS lines in recipe, by replacing them with PACKAGECONFIG

liuming50, 2019-01-30:
I have created a pull request probably can fix this issue: https://github.com/advancedtelematic/meta-updater/pull/468/commits/6c391d6856b0a831e9e62074ef01f184e305046e

agners, 2019-02-19:
@fernandoendo @patrickvacek with packageconfig merged (pull request #470) I think this issue is addressed.

patrickvacek, 2020-07-30:
Seems to have been fixed.

closed 2020-07-30 by patrickvacek