advancedtelematic/meta-updater #420: aktualizr fails when /var/sota/gateway.url is terminated by a newline

Bubu, 2018-10-31:
This is really unexpected behaviour, creating `/var/sota/gateway.url` as a normal file terminates it with a newline acording to POSIX. aktualizr makes that newline part of the url handed to curl, which complains about illegal characters.

patrickvacek, 2018-11-01:
Yes, looks like a bug. We normally generate that file automatically with `cert_provider`, but `aktualizr` really should ignore the newlines if found. In the meantime, you can always create the file with `echo -n` to avoid the newline.

patrickvacek, 2018-11-01:
@Bubu Thanks for catching this. It's been fixed in aktualizr. Until we bump the aktualizr version in the branch of meta-updater you use, you can use one of the standard tricks to get a newer version of aktualizr in your builds if you need to: https://github.com/advancedtelematic/meta-updater/#development-configuration

closed 2018-11-01 by patrickvacek