advancedtelematic/meta-updater #449: Checksum not calculated for initramfs and devicetree

LukaPivk, 2019-01-03:
Hi,

We ware reading ostree documentation and came across this statement 

>  If an initramfs or devicetree exist, the checksum must include all of the kernel, initramfs and devicetree contents
On this website https://ostree.readthedocs.io/en/latest/manual/deployment/

I see that image ostree only calculates kernel checksum and not device tree and initramfs. 
Is there a reason for that? This ostree change and meta-updater was not updated or is it simply a bug?
If its simply a bug I will create a pull request but I wanted to know if there is some reason behind it.

Thanks

patrickvacek, 2019-01-07:
@LukaPivk no reason that I am aware of. It looks like a bug to me. I think we either overlooked that line or it was added after we implemented the checksum and we never went back and updated it.

Good catch!

LukaPivk, 2019-01-07:
ok good. I will fix it test it and create pull request.
Thanks