advancedtelematic/meta-updater #460: GPIO Pin Modes

KarlEaves, 2019-01-24:
Hey guys, I'm trying to set up a custom-dts for my raspberry pi. I was able to get i2c working using 
`+
+&i2c1 {
+    #address-cells = <1>;
+    #size-cells = <0>;
+    status = "okay";
+
+    pcf8523: pcf8523@68 {
+        compatible = "nxp,pcf8523";
+        reg = <0x68>;
+        status = "okay";
+    };
+};`

Now I am trying to get UART working, which usually requires dtoverlay=pi3-disable-bt. Therefore, I added
`+&uart0{
+   pinctrl-names = "default";
+   pinctrl-0 = <&uart0_pins>;    
+   status = "okay";
+};
+
+&uart0_pins{
+   brcm,pins;
+   brcm,function;
+   brcm,pull;
+};
+
+&uart1{
+   status = "disabled";
+};`

However, using gpio allreadall, I see that the pins are all alternating in states ALT1 Value High and ALT5 Value Low. This is appearing with or without my dtb, and I'm wondering if there is anything you guys know that could be causing this? 

I am compiling on Ubuntu 16 for Raspberry Pi 3B+ and including the meta-updater as well as meta-updater-raspberrypi. 

My local.conf: 
`LICENSE_FLAGS_WHITELIST = "commercial"
DISTRO_FEATURES = "ext2 usbhost ${DISTRO_FEATURES_LIBC} systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "pulseaudio"
PREFERRED_PROVIDER_jpeg = "libjpeg-turbo"

PREFERRED_PROVIDER_jpeg-native = "libjpeg-turbo-native"
PREFERRED_PROVIDER_udev = "eudev"
VIRTUAL-RUNTIME_init_manager = "systemd"
MACHINE_FEATURES_remove = "apm"

PACKAGE_CLASSES = "package_ipk"
DISTRO = "poky-sota"
MACHINE = "raspberrypi3"
RPI_USE_U_BOOT = "1"
KERNEL_IMAGETYPE = "uImage"
KERNEL_MODULE_AUTOLOAD += " \
                        i2c-dev \
                        i2c-2708 \
                        rtc-pcf8523 \
                        "
KERNEL_DEVICETREE_raspberrypi3_sota = " \
                    bcm2710-rpi-3-b-plus-wispr.dtb \
                    "

DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
ENABLE_I2C = "1"
SERIAL_CONSOLE = "57600 ttyAMA0"
VIDEO_CAMERA = "1"

INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P wisprsystems root; "
USER_CLASSES = "image-mklibs image-prelink"
PATCHRESOLVE = "noop"
INHERIT += "rm_work"
CONF_VERSION = "1"`

Thanks,
Karl Eaves

KarlEaves, 2019-01-24:
I've also read the the device-tree-overlays are the last thing to affect the kernel. So I tried setting all the pins to IN in the config.txt, which would normally work. However, even without my overlay, something is still setting these pins to ALT1 and ALT5.

An example of my output from gpio allreadall:

|--------|-----------|----------|
| Pin   |  Mode  |  Value |
|--------|-----------|----------|
|---0---|-ALT1---|--HIGH-|
|---1---|-ALT5---|--LOW-|
|---2---|-ALT1---|--HIGH-|
|---3---|-ALT5---|--Low-|

This pattern continues throughout except for pin 19 which is set to IN and pin 43 which is set to ALT2. Normally all of these pins would be IN.

OYTIS, 2019-01-25:
Hi @KarlEaves ,
that sounds strange indeed. Could you also share the local.conf file that works (i.e. without meta-updater)? Also what recipe do you build: core-image-minimal, rpi-basic-image or something else?

Thanks!

KarlEaves, 2019-01-25:
Sure, here is my working local.conf

************************************************************************************
LICENSE_FLAGS_WHITELIST = "commercial"
DISTRO_FEATURES = "ext2 usbhost ${DISTRO_FEATURES_LIBC} systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "pulseaudio"
PREFERRED_PROVIDER_jpeg = "libjpeg-turbo"

PREFERRED_PROVIDER_jpeg-native = "libjpeg-turbo-native"
PREFERRED_PROVIDER_udev = "eudev"
VIRTUAL-RUNTIME_init_manager = "systemd"
MACHINE_FEATURES_remove = "apm"

KERNEL_MODULE_AUTOLOAD += " i2c-dev"

MACHINE = "raspberrypi3"
KERNEL_IMAGETYPE = "zImage"

DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
ENABLE_I2C = "1"
VIDEO_CAMERA = "1"

DISTRO = "poky"
PACKAGE_CLASSES = "package_ipk"
SDKMACHINE = "x86_64"
INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P wisprsystems root; "
USER_CLASSES = "image-mklibs image-prelink"
PATCHRESOLVE = "noop"
INHERIT += "rm_work"
CONF_VERSION = "1"
************************************************************************************

I am building a custom image off of console-image.bb, I have pasted it below

************************************************************************************
require ../../meta-rpi/images/console-image.bb

NODE_DEPS = " \
    modulecomms \
    add-cors-to-couchdb \
    bcrypt \
    better-sqlite3 \
    body-parser \
    chalk \
    events \
    express \
    fs-extra \
    http \
    jsonwebtoken \
    mkdirp \
    nodejs \
    path \
    request \
    request-promise \
    simple-peer \
    sleep \
    socket.io \
    socket.io-client \
    "

SYSTEM_APPS = " \
    logger \
    network-manager \
    flight-controller \
    auth \
    angularcom \
    mjpg \
    sqlite3 \
    "

SYSTEM_DEPS = " \
    busybox \
    parted \
    mjpg-streamer \
    wiringpi \
    "

NETWORK_PKGS = " \
    dhcpcd \
    iproute2 \
    dnsmasq \
    hostapd \
    wireless-tools \
    iw \
    linux-firmware-ralink \
    "

FCS_DEPS = " \
    libssh \
    libssl \
    boost \
    libtinyxml2 \
    libusb1-dev \
    wget \
    unzip \
    zlib \
    dronecode-dev \
    "

IMAGE_INSTALL += " \
    ${NODE_DEPS} \
    ${SYSTEM_APPS} \
    ${NETWORK_PKGS} \
    ${FCS_DEPS} \
    ${SYSTEM_DEPS} \
    "

IMAGE_INSTALL_remove += " \
    init-ifupdown \
    connman \
    connman-client \
    "

set_nodejs_path() {
  echo NODE_PATH=/usr/lib/node_modules > ${IMAGE_ROOTFS}/etc/environment
}

configure_network() {
  echo net.ipv4.ip_forward=1 > ${IMAGE_ROOTFS}/etc/sysctl.conf
  rm -rf ${IMAGE_ROOTFS}/etc/systemd/system/multi-user.target.wants/dnsmasq.service
  rm -rf ${IMAGE_ROOTFS}/etc/systemd/system/multi-user.target.wants/hciuart.service
}

set_kernel_log_level() {
  echo kernel.printk=3 4 1 3 > ${IMAGE_ROOTFS}/etc/sysctl.conf
}

set_gpu_mem() {
    sed -i '/#gpu_mem=/ c\gpu_mem=${GPU_MEM}' ${DEPLOYDIR}/bcm2835-bootfiles/config.txt
}

ROOTFS_POSTPROCESS_COMMAND += " \
    set_nodejs_path ; \
    set_kernel_log_level ; \
    configure_network ; \
 "

MACHINE_FEATURES_BACKFILL = "rtc"

IMAGE_ROOTFS_EXTRA_SPACE = "10000"

export IMAGE_BASENAME = "wispros"
************************************************************************************

I have tried it both as a zImage and uImage and it has the same strange behavior. I think I mentioned up there as well that I have also taken my patch out completely, and verified this because my i2cdetect no longer works, but the pins still alternate between ALT1 and ALT5. This is definitely a mind wrinkler!

OYTIS, 2019-01-28:
Hi @KarlEaves ,

thank you for all the details. It would really help us to have some minimal (as far as possible) image where this error is reproduced. I can't build the image recipe you've posted, because it contains some apparently private recipes (starting with `modulecomms`). So yes, if you could strip down everything unrelated, check that the problem is still there and then post your local.conf, bblayers.conf and the image recipe (or better you layer stripped of sensitive data), we could try to move forward on that.

Thanks!

KarlEaves, 2019-01-28:
I was able to reproduce when building console-image from meta-rpi. 

My local.conf is:
LICENSE_FLAGS_WHITELIST = "commercial"
DISTRO_FEATURES = "ext2 usbhost ${DISTRO_FEATURES_LIBC} systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "pulseaudio"
PREFERRED_PROVIDER_jpeg = "libjpeg-turbo"

PREFERRED_PROVIDER_jpeg-native = "libjpeg-turbo-native"
PREFERRED_PROVIDER_udev = "eudev"
VIRTUAL-RUNTIME_init_manager = "systemd"
MACHINE_FEATURES_remove = "apm"

PACKAGE_CLASSES = "package_ipk"
DISTRO = "poky-sota"
MACHINE = "raspberrypi3"
RPI_USE_U_BOOT = "1"
KERNEL_IMAGETYPE = "uImage"
# KERNEL_MODULE_AUTOLOAD += " \
#                         i2c-dev \
#                         i2c-2708 \
#                         rtc-pcf8523 \
#                         "

DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
ENABLE_I2C = "1"
SERIAL_CONSOLE = "57600 ttyAMA0"
VIDEO_CAMERA = "1"

INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P wisprsystems root; "
USER_CLASSES = "image-mklibs image-prelink"
PATCHRESOLVE = "noop"
INHERIT += "rm_work"
CONF_VERSION = "1"

bblayers.conf:
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

YOCTOROOT = "${@os.path.abspath(os.path.join("${TOPDIR}", os.pardir))}"

BBLAYERS ?= " \
  ${YOCTOROOT}/meta \
  ${YOCTOROOT}/meta-poky \
  ${YOCTOROOT}/meta-wispros \
  ${YOCTOROOT}/meta-raspberrypi \
  ${YOCTOROOT}/meta-rpi \
  ${YOCTOROOT}/meta-qt5 \
  ${YOCTOROOT}/meta-security \
  ${YOCTOROOT}/meta-openembedded/meta-python \
  ${YOCTOROOT}/meta-openembedded/meta-networking \
  ${YOCTOROOT}/meta-openembedded/meta-oe \
  ${YOCTOROOT}/meta-openembedded/meta-perl \
  ${YOCTOROOT}/meta-openembedded/meta-multimedia \
  ${YOCTOROOT}/meta-openembedded/meta-filesystems \
  ${YOCTOROOT}/meta-maker \
  ${YOCTOROOT}/meta-updater \
  ${YOCTOROOT}/meta-updater-raspberrypi \
  "

image recipe:
SUMMARY = "A console development image with some C/C++ dev tools"
HOMEPAGE = "http://www.jumpnowtek.com"
LICENSE = "MIT"

IMAGE_FEATURES += "package-management splash"
IMAGE_LINGUAS = "en-us"

inherit image

DEPENDS += "bcm2835-bootfiles"

CORE_OS = " \
    kernel-modules \
    openssh openssh-keygen openssh-sftp-server \
    packagegroup-core-boot \
    term-prompt \
    tzdata \
"

WIFI_SUPPORT = " \
    crda \
    iw \
    linux-firmware-raspbian \
    wpa-supplicant \
"

DEV_SDK_INSTALL = " \
    binutils \
    binutils-symlinks \
    coreutils \
    cpp \
    cpp-symlinks \
    diffutils \
    file \
    g++ \
    g++-symlinks \
    gcc \
    gcc-symlinks \
    gdb \
    gdbserver \
    gettext \
    git \
    ldd \
    libstdc++ \
    libstdc++-dev \
    libtool \
    make \
    pkgconfig \
    python3-modules \
"

DEV_EXTRAS = " \
    ntp \
    ntp-tickadj \
    serialecho  \
    spiloop \
"

EXTRA_TOOLS_INSTALL = " \
    bzip2 \
    devmem2 \
    dosfstools \
    ethtool \
    fbset \
    findutils \
    i2c-tools \
    iperf3 \
    iproute2 \
    iptables \
    less \
    nano \
    netcat \
    procps \
    rndaddtoentcnt \
    rng-tools \
    sysfsutils \
    tcpdump \
    unzip \
    util-linux \
    wget \
    zip \
"

RPI_STUFF = " \
    omxplayer \
    raspi2fb \
    userland \
    wiringpi \
"

IMAGE_INSTALL += " \
    ${CORE_OS} \
    ${DEV_SDK_INSTALL} \
    ${DEV_EXTRAS} \
    ${EXTRA_TOOLS_INSTALL} \
    ${RPI_STUFF} \
    ${WIFI_SUPPORT} \
"

set_local_timezone() {
    ln -sf /usr/share/zoneinfo/EST5EDT ${IMAGE_ROOTFS}/etc/localtime
}

disable_bootlogd() {
    echo BOOTLOGD_ENABLE=no > ${IMAGE_ROOTFS}/etc/default/bootlogd
}

ROOTFS_POSTPROCESS_COMMAND += " \
    set_local_timezone ; \
    disable_bootlogd ; \
 "

export IMAGE_BASENAME = "console-image"

I left everything the same in the console-image, besides adding in wiringpi so that I could check the gpio states. I still have the same error, where the pins all alternate between ALT1 and ALT5.

OYTIS, 2019-01-30:
@KarlEaves @patrickvacek @eu-smirnov @lbonn 

For now I can confirm that the problem is reproduced in the very minimal yocto environment without meta-rpi (as created by [updater-repo](https://github.com/advancedtelematic/updater-repo) ) and building rpi-basic-image on `thud`.

The problem looks serious and needs to be fixed sooner rather than later. I'm trying to get more info.

OYTIS, 2019-01-31:
OK, so looks like the problem itself is not that serious (at least we're not corrupting any registers), but the solution might be a bit tricky.

The `gpio` tool looks into /proc/cpuinfo to determine the base address for peripheral registers. Namely what is important is Revision field. The Revision field is in turn based on /system/linux,revision node of the device tree. This node is not present in *.dtb file compiled with the linux kernel, but rather is automagically added by a proprietary Raspberry Pi bootloader.

Now the problem is that because we want device tree and the overlays to be updateable, we can't use the device tree that Raspberry Pi bootloader produces: it's based on a file in the boot partition and is not controlled by OSTree. Instead of that meta-updater packs the chosen device tree and overlays into a FIT image on the build time.

Obviously, this image doesn't contain the /system/linux,revision field, so the revision in /proc/cpuinfo stays zero, and `gpio` uses wrong peripheral address when displaying GPIO information. You can still use /sys/gpio without any problem.

The proper solution would be to have device tree overlays directly supported by OSTree, and then we can stop using FIT images for Rasperry Pi and manipulate the device tree from u-boot. I'm not sure how much time that can take.

If you're just concerned about the `gpio` tool, we can add a patch for it to meta-updater-raspberrypi to ignore the board revision. If there's something else that doesn't work for you because of that, please describe in more detail and we'll see what can be done.

ricardosalveti, 2019-01-31:
> The proper solution would be to have device tree overlays directly supported by OSTree, and then we can stop using FIT images for Rasperry Pi and manipulate the device tree from u-boot.

You can still use FIT and retrieve the device tree from u-boot, but then you might have update issues later on in case the device tree also needs to be updated. Wouldn't just be easier to add this extra overlay as part of the fitimage and load it during runtime?

An example of how we're loading overlays during boot:
```
$ cat uEnv.txt
dtoverlays=#conf@overlays_vc4-kms-v3d.dtbo
bootcmd_otenv=load ${devtype} ${devnum}:2 ${loadaddr} /boot/loader/uEnv.txt; env import -t ${loadaddr} ${filesize}
bootcmd_load_f=load ${devtype} ${devnum}:2 ${ramdisk_addr_r} "/boot"${kernel_image}
bootcmd_run=setexpr fdtfile gsub '/' '_'; bootm ${ramdisk_addr_r}#conf@${fdtfile}${dtoverlays}
bootcmd=run bootcmd_otenv; run bootcmd_load_f; run bootcmd_run
```

With this setup the user just needs to include the desired device tree overlay as part of the build process (in order to be included in the fitimage) and change the uEnv.txt file that is available in the first partition (you can enable more overlays by appending #conf@overlays_name.dtbo at dtoverlays)

KarlEaves, 2019-01-31:
@OYTIS Thanks for the insight! So gpio is displaying incorrectly? I was only using it to verify that my pins were set up for UART serial communication, which is  my biggest issue. If I can get the serial ports to work correctly I do not need gpio, as it was just a tool to verify.
I'm trying to set up the raspberry pi to communicate with an external flight-controller. I do this by setting it up to transfer data over ttyAMA0 (UART pins 8&10/15&16) with a baud rate of 57600. I can usually verify that the Pi is receiving info by running cat /dev/ttyAMA0, but this is not returning any info after adding meta-updater/meta-updater-raspberrypi/u-boot, likely because I am confused somewhere. 
I know this may be hard to test without a device that communicates over the serial port, but here is my build set up with most everything stripped out:

```
wispros.bb

require ../../meta-rpi/images/console-image.bb
export IMAGE_BASENAME = "wispros"

```
 
```
layer.conf

# We have a conf and classes directory, add to BBPATH
BBPATH .= ":${LAYERDIR}"

#We have recipes-* directories, add to BBFILES
BBFILES += "${LAYERDIR}/recipes-*/*/*.bb \
            ${LAYERDIR}/recipes-*/*/*.bbappend \
            ${LAYERDIR}/images/*.bb"

BBFILE_COLLECTIONS += "meta-wispros"
BBFILE_PATTERN_meta-wispros = "^${LAYERDIR}/"
BBFILE_PRIORITY_meta-wispros = "100"

LAYERSERIES_COMPAT_meta-wispros = "sumo"

```

And I'm including this patch
```
0001-custom-dts.patch
From 8582311b1921f5a6de4322531c141a93f902637d Mon Sep 17 00:00:00 2001
From: Mike Allen <mallen3699@gmail.com>
Date: Fri, 4 Jan 2019 13:02:51 -0600
Subject: [PATCH] custom dts

---
 .../boot/dts/bcm2710-rpi-3-b-plus-wispr.dts   | 27 +++++++++++++++++++
 1 file changed, 27 insertions(+)
 create mode 100644 arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts

diff --git a/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts b/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts
new file mode 100644
index 000000000000..1e26363b3eed
--- /dev/null
+++ b/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts
@@ -0,0 +1,43 @@
+/* Custom dts file for raspberry pi 3 b plus for WISPr Systems */
+
+#include "bcm2710-rpi-3-b-plus.dts"
+
+&i2c1 {
+    #address-cells = <1>;
+    #size-cells = <0>;
+    status = "okay";
+
+    pcf8523: pcf8523@68 {
+        compatible = "nxp,pcf8523";
+        reg = <0x68>;
+        status = "okay";
+    };
+};
+
+&uart0{
+   pinctrl-names = "default";
+   pinctrl-0 = <&uart0_pins>;    
+   status = "okay";
+};
+
+&uart0_pins{
+   brcm,pins;
+   brcm,function;
+   brcm,pull;
+};
+
+&uart1{
+   status = "disabled";
+};
+
+/ {
+    __overrides__ {
+        pcf8523 = <0>,"+7";
+        addr = <&pcf8523>, "reg:0";
+    };
+
+    aliases {
+        serial0 = "/soc/serial@7e201000";
+        serial1 = "/soc/serial@7e215040";
+    };
+};
\ No newline at end of file
-- 
2.17.1

```

This works for the i2c patch but not the uart. 
@ricardosalveti I will try adding the overlay during boot instead of creating a custom dts but I'm not sure how to do this yet. I will look into it and respond if it helps! 
Thanks

OYTIS, 2019-01-31:
@ricardosalveti 
That `#conf@` trick looks pretty cool, how does that work? I couldn't find it documented anywhere.

Then yes, we could have a number of dtbos in the boot partition and load one or another depending on what's in fdt provided by the primary bootloader. It would be even better to provide the serial/revision from the primary bootloader as it is, but for that we need to be able to manipulate the resulting dtb from the u-boot script, which we can't do to my knowledge.

ricardosalveti, 2019-01-31:
> That #conf@ trick looks pretty cool, how does that work? I couldn't find it documented anywhere.

That's how overlays are handled by u-boot, have a look at https://github.com/u-boot/u-boot/blob/master/doc/uImage.FIT/overlay-fdt-boot.txt. Since OE already includes all the overlays specified in KERNEL_DEVICETREE (as an extra configuration), all you need to do is tell u-boot to load the overlay you want.

ricardosalveti, 2019-01-31:
>                default = "conf@bcm2710-rpi-3-b-plus-wispr.dtb";

Here your changes are part of the default device-tree that the board will use when booting (default configuration), so you shouldn't need any extra change in uEnv.txt (since the default configuration is the one selected during boot).

If you want to have your extra changes as an overlay, you would need to do it similar to https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/overlays/pi3-miniuart-bt-overlay.dts and then change KERNEL_DEVICETREE_raspberrypi3_sota to include it as part of the build (e.g. KERNEL_DEVICETREE_raspberrypi3_sota ?= " bcm2710-rpi-3-b.dtb overlays/vc4-kms-v3d.dtbo overlay/your-overlay.dtbo).

ricardosalveti, 2019-01-31:
> It would be even better to provide the serial/revision from the primary bootloader as it is, but for that we need to be able to manipulate the resulting dtb from the u-boot script, which we can't do to my knowledge.

I was thinking if it would be possible to have the serial/revision update piece done by u-boot when loading and manipulating the device-tree, like how it is already done for simplefb: https://github.com/qemu/u-boot/blob/master/board/raspberrypi/rpi/rpi.c#L478. There is already code to set the board revision and name via environment variables (https://github.com/qemu/u-boot/blob/master/board/raspberrypi/rpi/rpi.c#L350), we would just need to have that as part of the device-tree instead.

KarlEaves, 2019-01-31:
I can verify that it is indeed using my custom dts, but for some reason the UART changes aren't going into affect.

This is the pi3-disable-bt-overlay.dts that is usually required to enable serial/uart.
```
/dts-v1/;
/plugin/;

/* Disable Bluetooth and restore UART0/ttyAMA0 over GPIOs 14 & 15.
   To disable the systemd service that initialises the modem so it doesn't use
   the UART:

       sudo systemctl disable hciuart
*/

/{
	compatible = "brcm,bcm2708";

	fragment@0 {
		target = <&uart1>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@1 {
		target = <&uart0>;
		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&uart0_pins>;
			status = "okay";
		};
	};

	fragment@2 {
		target = <&uart0_pins>;
		__overlay__ {
			brcm,pins;
			brcm,function;
			brcm,pull;
		};
	};

	fragment@3 {
		target-path = "/aliases";
		__overlay__ {
			serial0 = "/soc/serial@7e201000";
			serial1 = "/soc/serial@7e215040";
		};
	};
};
```
and thats what I add to my custom dts
```
0001-custom-dts.patch
From 8582311b1921f5a6de4322531c141a93f902637d Mon Sep 17 00:00:00 2001
From: Mike Allen <mallen3699@gmail.com>
Date: Fri, 4 Jan 2019 13:02:51 -0600
Subject: [PATCH] custom dts

---
 .../boot/dts/bcm2710-rpi-3-b-plus-wispr.dts   | 27 +++++++++++++++++++
 1 file changed, 27 insertions(+)
 create mode 100644 arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts

diff --git a/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts b/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts
new file mode 100644
index 000000000000..1e26363b3eed
--- /dev/null
+++ b/arch/arm/boot/dts/bcm2710-rpi-3-b-plus-wispr.dts
@@ -0,0 +1,43 @@
+/* Custom dts file for raspberry pi 3 b plus for WISPr Systems */
+
+#include "bcm2710-rpi-3-b-plus.dts"
+
+&i2c1 {
+    #address-cells = <1>;
+    #size-cells = <0>;
+    status = "okay";
+
+    pcf8523: pcf8523@68 {
+        compatible = "nxp,pcf8523";
+        reg = <0x68>;
+        status = "okay";
+    };
+};
+
+&uart0{
+   pinctrl-names = "default";
+   pinctrl-0 = <&uart0_pins>;    
+   status = "okay";
+};
+
+&uart0_pins{
+   brcm,pins;
+   brcm,function;
+   brcm,pull;
+};
+
+&uart1{
+   status = "disabled";
+};
+
+/ {
+    __overrides__ {
+        pcf8523 = <0>,"+7";
+        addr = <&pcf8523>, "reg:0";
+    };
+
+    aliases {
+        serial0 = "/soc/serial@7e201000";
+        serial1 = "/soc/serial@7e215040";
+    };
+};
\ No newline at end of file
-- 
2.17.1
```

And I can verify that this is enabling i2c (the top part of the dts works), but I can not verify that it is enabling UART/disabling UART1. I would normally use gpio to test that UART was enabled, but its using the wrong peripheral address so it is not displaying correctly. Is there way to test this other than gpio?
Also, does making these changes an overlay affect the board differently than making the custom dts? For example, could the pins be set incorrectly because of another overlay affecting the board after it has already loaded my custom dts?

KarlEaves, 2019-01-31:
Btw, this is all amazing info! Thanks for being so quick to respond

OYTIS, 2019-02-01:
OK, so looks like support for device tree overlays in FIT images is missing from our u-boot scripts altogether. We'll think how to integrate it better with OSTree, probably some snippets on the deployed file system will work.

And setting the board revision in u-boot code seems also the right thing to do to me.

Thank you!

KarlEaves, 2019-02-05:
@OYTIS I am still having trouble getting the UART to work correctly on ttyAMA0. I have tried patching the uboot source so that it doesn't change the gpio pin settings while also patching bcm2710-rpi-3-b, but for some reason the ttyS0 is still being used instead of AMA0. 

I am on a tight schedule right now and am worried I won't be able to get this to work in time. Is there a way that I can get overlays to work again by reverting to how it was done before FIT images while still being able to update everything besides the hardware layer? (a bad idea, I know. Eventually we will integrate the full kernel updates with our system)

ricardosalveti, 2019-02-06:
I wonder if your final device-tree is still missing something that would usually be populated by the firmware (I know uart gets manipulated by the firmware depending on the rpi rev, like flow control). One easy way to debug is just dumping the device tree from your running system and compare with a similar modification but done with raspbian.

You can dump the running device-tree by executing `dtc -I fs /proc/device-tree`

OYTIS, 2019-02-06:
Hi @KarlEaves, @ricardosalveti,

we've added some patches to meta-updater and meta-updater-raspberrypi. With the master branch of both you should be able to add device tree overlays to you image. The overlays should first be listed in `KERNEL_DEVICETREE` so that they are packed into the FIT image, and then in `SOTA_DT_OVERLAYS` to have them loaded by the bootloader. They are given to these lists in a bit different manner: `KERNEL_DEVICETREE` is a poky variable and requires the relative path in the Linux build directory while `SOTA_DT_OVERLAYS` is defined in meta-updater and requires just a filename. See the [RPi bbclass in meta-updater](https://github.com/advancedtelematic/meta-updater/blob/master/classes/sota_raspberrypi.bbclass) for the example of both.
Thank you for helping us find that problem and looking forward to your feedback.

KarlEaves, 2019-02-06:
@OYTIS I am updating all my layers to thud now so that it is compatible and will test it shortly. 
 Thanks!

KarlEaves, 2019-02-06:
@ricardosalveti 
I tried running `dtc -I fs /proc/device-tree` but it returned multiple warnings saying that "cell * is not a phandle reference", then threw a segmentation fault. Could this be because of something I'm doing wrong?

@OYTIS I am able to build the image with the following local.conf and `bitbake console-image`

```
LICENSE_FLAGS_WHITELIST = "commercial"
DISTRO_FEATURES = "ext2 usbhost ${DISTRO_FEATURES_LIBC} systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "pulseaudio"
PREFERRED_PROVIDER_jpeg = "libjpeg-turbo"

PREFERRED_PROVIDER_jpeg-native = "libjpeg-turbo-native"
PREFERRED_PROVIDER_udev = "eudev"
VIRTUAL-RUNTIME_init_manager = "systemd"
MACHINE_FEATURES_remove = "apm"

PACKAGE_CLASSES = "package_ipk"
DISTRO = "poky-sota"
MACHINE = "raspberrypi3"
RPI_USE_U_BOOT = "1"

KERNEL_DEVICETREE_raspberrypi3 = "bcm2710-rpi-3-b.dtb overlays/pi3-disable-bt.dtbo overlays/i2c-rtc.dtbo"

SOTA_MAIN_DTB_raspberrypi3 = "bcm2710-rpi-3-b.dtb"

SOTA_DT_OVERLAYS_raspberrypi3 = "pi3-disable-bt.dtbo i2c-rtc.dtbo"

DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
ENABLE_I2C = "1"
SERIAL_CONSOLE = "57600 ttyAMA0"
VIDEO_CAMERA = "1"

INSANE_SKIP_${PN}-dev += "dev-elf"
FILES_${PN}-dev += "${libdir}/libaws-cpp-sdk-core.so ${libdir}/libaws-cpp-sdk-s3.so"
FILES_${PN} += "${libdir}/libaws-cpp-sdk-core.so ${libdir}/libaws-cpp-sdk-s3.so"

INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P wisprsystems root; "
USER_CLASSES = "image-mklibs image-prelink"
PATCHRESOLVE = "noop"
INHERIT += "rm_work"
CONF_VERSION = "1"
```
However, after running `i2cdetect -y 1` it returns `Error: Could not open file '/dev/i2c-1': No such file or directory` which usually what happens when the overlay is not applied, and the UART is still printing on ttyS0 instead of ttyAMA0. Am I be missing something else in the build?

ricardosalveti, 2019-02-07:
> I tried running `dtc -I fs /proc/device-tree` but it returned multiple warnings saying that "cell * is not a phandle reference", then threw a segmentation fault. Could this be because of something I'm doing wrong?

I believe this is just because the distribution got an older version of the device tree compiler, and you can probably get around that by compiling your own (if you want to go that path). This might be handy anyway as you might find similar issues even when doing all your overlay work via fitimages, as the proprietary firmware is known to modify a few things during runtime.

OYTIS, 2019-02-07:
For dtc you can pull [this PR](https://github.com/advancedtelematic/meta-updater-raspberrypi/pull/47), it is indeed broken in poky master.

For these specific overlays, I'm building the image right now to understand where the problem can be.

OYTIS, 2019-02-07:
I've tried both of these overlays so far, and the second one, `i2c-rtc.dtbo` would prevent the image from booting. So if it does boot for you, please re-check that you have the latest `master` version of both `meta-updater` and `meta-updater-rasbperrypi` =) I suspect it wouldn't apply because it wants at least one parameter to be passed to it, but that's just a wild guess, I didn't dig deep yet. Also I couldn't find a way yet to pass 'dtparams' with u-boot, it might be a blocker, but I hope it's not, will keep you updated.

As for `pi3-disable-bt`, it seems to work, ttyS0 really disappears, ttyAMA0 really appears. Again, please check that both layers are on master and don't forget to redefine `OSTREE_KERNEL_ARGS_sota` to use ttyAMA0 as a serial console. Default value for `OSTREE_KERNEL_ARGS_sota` can be seen [here](https://github.com/advancedtelematic/meta-updater/blob/8c10b573a9dc000899fd71d3bd0d82131200ac23/classes/sota_raspberrypi.bbclass#L34).

KarlEaves, 2019-02-07:
For some reason I'm having to change 0001-board-raspberry-add-serial-revision-to-the-dev.patch to this 

```
diff --git a/board/raspberrypi/rpi/rpi.c b/board/raspberrypi/rpi/rpi.c
index 3b7a54f519..11f7c14749 100644
--- a/board/raspberrypi/rpi/rpi.c
+++ b/board/raspberrypi/rpi/rpi.c
@@ -205,6 +205,8 @@ static uint32_t rev_scheme;
 static uint32_t rev_type;
 static const struct rpi_model *model;
 
+uint64_t serial;
+
 #ifdef CONFIG_ARM64
 static struct mm_region bcm2837_mem_map[] = {
 	{
@@ -347,9 +349,9 @@ static void set_serial_number(void)
 		/* Ignore error; not critical */
 		return;
 	}
-
-	snprintf(serial_string, sizeof(serial_string), "%016" PRIx64,
-		 msg->get_board_serial.body.resp.serial);
+    serial = msg->get_board_serial.body.resp.serial;
+	snprintf(serial_string, sizeof(serial_string), "%016llx", serial);
+	
 	env_set("serial#", serial_string);
 }
 
@@ -485,6 +487,30 @@ void *board_fdt_blob_setup(void)
 	return (void *)fw_dtb_pointer;
 }
 
+ 
+static int ft_add_revision_info(void *blob) {
+	int off;
+	int ret;
+
+	off = fdt_subnode_offset(blob, 0, "system");
+
+	if (off < 0) {
+		off = fdt_add_subnode(blob, 0, "system");
+		if (off < 0)
+			return -1;
+	}
+
+	ret = fdt_setprop_u64(blob, off, "linux,serial", serial);
+	if (ret < 0)
+		return -1;
+
+	ret = fdt_setprop_u32(blob, off, "linux,revision", revision);
+	if (ret < 0)
+		return -1;
+
+	return 0;
+}
+
 int ft_board_setup(void *blob, bd_t *bd)
 {
 	/*
@@ -493,6 +519,7 @@ int ft_board_setup(void *blob, bd_t *bd)
 	 * node exists for the "real" graphics driver.
 	 */
 	lcd_dt_simplefb_add_node(blob);
+    ft_add_revision_info(blob);
 
 #ifdef CONFIG_EFI_LOADER
 	/* Reserve the spin table */

```
Could we be on different versions of u-boot and would this cause it to be off? It still builds once I change it

OYTIS, 2019-02-07:
@KarlEaves Oh yes, these are differences between u-boot revisions in master in thud branches.

KarlEaves, 2019-02-07:
Can confirm that ttyS0 does not show anymore, and ttyAMA0 is there, but I'm still not receiving data from the flight controller. Looking into this now

OYTIS, 2019-02-07:
As for i2c-rtc overlay, there are two problems related to Raspberry Pi proprietary extensions to device tree format. First is `dtparams` and related `__overrides__` field. Unfortunately u-boot can't manage this, so the only way now is to create new device tree overlays manually applying the overrides.
Second, a minor one is that RPi's overlays use some magic node names, one of them being `i2c_arm`. In your custom overlay you should replace occurrences of `i2c_arm` with `i2c0` or `i2c1` at your choice.
Here's the example of an overlay for ds1307 RTC. I couldn't test it, because I don't have the hardware, but at least it applies:

```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";

	fragment@0 {
		target = <&i2c1>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			ds1307: ds1307@68 {
				compatible = "maxim,ds1307";
				reg = <0x68>;
				status = "okay";
			};
		};
	};
};
```

Also if you want to use user-space interface to i2c (`/dev/i2c-X`), you should `modprobe i2c-dev` first.

KarlEaves, 2019-02-07:
Sounds good, I was able to get i2c working with that method before so I will stick to that for now.
I'm still having issues with UART though! D:
If my telemetry cable is plugged in during boot, it interrupts the boot (which I have a patch for, but have not put it in until I can get it working without it). So for now I'm waiting until after boot to plug in the telemetry, then running `cat /dev/ttyAMA0`  but I am still not able to pull in data. I have tried masking the serial-getty@ttyAMA0 to see if that was the issue but it is still not pulling data. I'm wondering if something weird is happening during the boot with the GPIO pins? I am going to apply your dtc patch and check out the device-tree

KarlEaves, 2019-02-07:
After comparing both device-trees, I see some things that are likely causing the issue.

in the` __overides__ `section,
on a working image using raspbian :
```
uart0 = [00 00 00 1d 73 74 61 74 75 73 00]
uart1 = [00 00 00 1e 73 74 61 74 75 73 00]
```

on my embedded system:
```
uart0 = [00 00 00 1e 73 74 61 74 75 73 00]
uart1 = [00 00 00 1f 73 74 61 74 75 73 00]
```

However, under the 
```
soc{
  serial@7e201000{
```
on a working image using raspbian :
`phandle = {0x1d}
`on my embedded system:
`phandle = {0x1e}`

KarlEaves, 2019-02-07:
It looks like gpio is now displaying the correct data. Pins  14 and 15 are in ALT5 (UART1) instead of ALT0 (UART0). If I use gpio to change them to ALT0 `gpio -g mode 14 ALT0` cat /dev/ttyAMA0 displays the data!!!! Great to see! Now to figure out why they're being put into ALT5 instead of ALT0...

But I can definitely get this to work now. Thanks for all your help!

KarlEaves, 2019-02-11:
@OYTIS I'm now getting this error when trying to build. I'm not sure exactly what is happening and am having trouble finding any insight online. Do you know what might be causing this?

```
ERROR: initramfs-ostree-image-1.0-r0 do_rootfs: The postinstall intercept hook 'update_font_cache' failed, details in /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs
ERROR: initramfs-ostree-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs.32691
ERROR: Task (/home/karleaves/Desktop/updater/poky/meta-updater/recipes-core/images/initramfs-ostree-image.bb:do_rootfs) failed with exit code '1'
ERROR: wispros-1.0-r0 do_rootfs: The postinstall intercept hook 'update_font_cache' failed, details in /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wispros/1.0-r0/temp/log.do_rootfs
ERROR: wispros-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/wispros/1.0-r0/temp/log.do_rootfs.32694
ERROR: Task (/home/karleaves/Desktop/updater/poky/meta-wispros/images/wispros.bb:do_rootfs) failed with exit code '1'
```

KarlEaves, 2019-02-11:
```
cat /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-core/glibc/cross-localedef-native_2.28.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['opkg-utils-native', 'depmodwrapper-cross', 'mklibs-native', 'opkg-native', 'makedevs-native', 'pseudo-native', 'update-rc.d-native', 'qemuwrapper-cross', 'pigz-native', 'ldconfig-native', 'cross-localedef-native', 'quilt-native', 'xz-native', 'openssl-native', 'debianutils-native', 'systemd-systemctl-native', 'libtool-native', 'autoconf-native', 'gettext-minimal-native', 'gnu-config-native', 'automake-native', 'shadow-native', 'pkgconfig-native', 'libsolv-native', 'libarchive-native', 'qemu-native', 'kmod-native', 'zlib-native', 'texinfo-dummy-native', 'm4-native', 'ninja-native', 'rpm-native', 'expat-native', 'cmake-native', 'e2fsprogs-native', 'lzo-native', 'bzip2-native', 'dtc-native', 'glib-2.0-native', 'pixman-native', 'alsa-lib-native', 'gtk-doc-native', 'python3-native', 're2c-native', 'elfutils-native', 'popt-native', 'file-native', 'db-native', 'dbus-native', 'nss-native', 'curl-native', 'util-linux-native', 'attr-native', 'flex-native', 'libpcre-native', 'gettext-native', 'libffi-native', 'util-macros-native', 'libpng-native', 'sqlite3-native', 'readline-native', 'gdbm-native', 'nspr-native', 'ncurses-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_rootfs
NOTE: Initializing intercept dir for /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/rootfs
DEBUG: Collected intercepts:
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/delay_to_first_boot
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/postinst_intercept
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_font_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_gio_module_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_gtk_immodules_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_icon_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_pixbuf_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_udev_hwdb

NOTE: ###### Generate rootfs #######
NOTE: Executing create_merged_usr_symlinks_rootfs ...
DEBUG: Executing shell function create_merged_usr_symlinks_rootfs
DEBUG: Shell function create_merged_usr_symlinks_rootfs finished
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4/Packages -p /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4/Packages -m /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4' ...
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all/Packages -p /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all/Packages -m /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all' ...
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/ [... cut ...]
Note that if you provide several changes to a single variable
the last change will stay in effect.

See <https://qemu.org/contribute/report-a-bug> for how to report bugs.
More information on the QEMU project at <https://qemu.org>.

ERROR: The postinstall intercept hook 'update_font_cache' failed, details in /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs
DEBUG: Python function do_rootfs finished
ERROR: Function failed: do_rootfs
karleaves@wisprdev-main:~/Desktop/
```

KarlEaves, 2019-02-11:
Here is the log file
```
cat /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs.26647
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/karleaves/Desktop/updater/poky/meta/recipes-core/glibc/cross-localedef-native_2.28.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/home/karleaves/Desktop/updater/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['cross-localedef-native', 'mklibs-native', 'pigz-native', 'pseudo-native', 'opkg-native', 'ldconfig-native', 'depmodwrapper-cross', 'opkg-utils-native', 'update-rc.d-native', 'makedevs-native', 'qemuwrapper-cross', 'xz-native', 'quilt-native', 'libtool-native', 'autoconf-native', 'gnu-config-native', 'automake-native', 'qemu-native', 'systemd-systemctl-native', 'debianutils-native', 'openssl-native', 'kmod-native', 'shadow-native', 'gettext-minimal-native', 'zlib-native', 'libsolv-native', 'pkgconfig-native', 'libarchive-native', 'texinfo-dummy-native', 'm4-native', 'dtc-native', 'glib-2.0-native', 'pixman-native', 'alsa-lib-native', 'gtk-doc-native', 'python3-native', 'cmake-native', 'ninja-native', 'rpm-native', 'expat-native', 'bzip2-native', 'e2fsprogs-native', 'lzo-native', 'flex-native', 'libffi-native', 'gettext-native', 'util-linux-native', 'libpcre-native', 'util-macros-native', 'libpng-native', 'gdbm-native', 'readline-native', 'sqlite3-native', 'curl-native', 're2c-native', 'nss-native', 'elfutils-native', 'popt-native', 'db-native', 'file-native', 'dbus-native', 'attr-native', 'ncurses-native', 'nspr-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_rootfs
NOTE: Initializing intercept dir for /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/rootfs
DEBUG: Collected intercepts:
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/delay_to_first_boot
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/postinst_intercept
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_font_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_gio_module_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_gtk_immodules_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_icon_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_pixbuf_cache
  /home/karleaves/Desktop/updater/poky/scripts/postinst-intercepts/update_udev_hwdb

NOTE: ###### Generate rootfs #######
NOTE: Executing create_merged_usr_symlinks_rootfs ...
DEBUG: Executing shell function create_merged_usr_symlinks_rootfs
DEBUG: Shell function create_merged_usr_symlinks_rootfs finished
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4/Packages -p /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4/Packages -m /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/cortexa7t2hf-neon-vfpv4' ...
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all/Packages -p /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all/Packages -m /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/oe-rootfs-repo/all' ...
NOTE: Executing '/home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/usr/bin/opkg-make-index -r /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnue [... cut ...]
    QEMU_SET_ENV=var1=val2,var2=val2 QEMU_UNSET_ENV=LD_PRELOAD,LD_DEBUG
Note that if you provide several changes to a single variable
the last change will stay in effect.

See <https://qemu.org/contribute/report-a-bug> for how to report bugs.
More information on the QEMU project at <https://qemu.org>.

ERROR: The postinstall intercept hook 'update_font_cache' failed, details in /home/karleaves/Desktop/updater/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs
DEBUG: Python function do_rootfs fin
```

KarlEaves, 2019-02-12:
Did a clean build and it worked. Not sure what the issue was there

KarlEaves, 2019-02-13:
I've run into the same error. It looks like it's after a run a git add and git commit on my repo. Very strange issue that's causing a lot of headaches and forcing me to build from scratch and not be able to update my repo.

OYTIS, 2019-02-14:
@KarlEaves 
Never seen that before unfortunately. Maybe the log has more info?

KarlEaves, 2019-02-14:
@OYTIS Here is the log
```
 cat /home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs.25725
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/karleaves/Desktop/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/poky/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', '/home/karleaves/Desktop/poky/meta/recipes-core/glibc/cross-localedef-native_2.28.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', '/home/karleaves/Desktop/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', '/home/karleaves/Desktop/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', '/home/karleaves/Desktop/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', 'virtual:native:/home/karleaves/Desktop/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/karleaves/Desktop/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['update-rc.d-native', 'opkg-native', 'makedevs-native', 'opkg-utils-native', 'pigz-native', 'mklibs-native', 'ldconfig-native', 'systemd-systemctl-native', 'libsolv-native', 'libarchive-native', 'shadow-native', 'debianutils-native', 'expat-native', 'rpm-native', 'ninja-native', 'cmake-native', 'e2fsprogs-native', 'nss-native', 'popt-native', 'dbus-native', 'file-native', 'db-native', 'elfutils-native', 're2c-native', 'curl-native', 'attr-native', 'nspr-native']
NOTE: Skipping as already exists in sysroot: ['cross-localedef-native', 'depmodwrapper-cross', 'pseudo-native', 'qemuwrapper-cross', 'xz-native', 'kmod-native', 'automake-native', 'autoconf-native', 'gnu-config-native', 'pkgconfig-native', 'libtool-native', 'quilt-native', 'qemu-native', 'zlib-native', 'gettext-minimal-native', 'openssl-native', 'python3-native', 'gtk-doc-native', 'texinfo-dummy-native', 'm4-native', 'bzip2-native', 'lzo-native', 'pixman-native', 'alsa-lib-native', 'glib-2.0-native', 'dtc-native', 'sqlite3-native', 'readline-native', 'gdbm-native', 'util-linux-native', 'util-macros-native', 'libpng-native', 'libffi-native', 'libpcre-native', 'gettext-native', 'flex-native', 'ncurses-native']
DEBUG: sed -e 's:^[^/]*/:/home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native/:g' /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/rpm-native/fixmepath /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/e2fsprogs-native/fixmepath /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/dbus-native/fixmepath /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/elfutils-native/fixmepath /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/curl-native/fixmepath /home/karleaves/Desktop/poky/build/tmp/sysroots-components/x86_64/nspr-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/recipe-sysroot-native:g' -e 's:FIXME_COMPONENTS_DIR:/home/karleaves/Desktop/poky/build/tmp/sysroots-components:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/karleaves/Desktop/poky/build/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/karleaves/Desktop/poky/build/tmp/pkgdata/raspberrypi3:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/fifo.25725:g'
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_rootfs
NOTE: Initializing intercept dir for /home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/rootfs
DEBUG: Collected intercepts:
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/delay_to_first_boot
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/postinst_intercept
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_font_cache
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_gio_module_cache
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_gtk_immodules_cache
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_icon_cache
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_pixbuf_cache
  /home/karleaves/Desktop/poky/scripts/postinst-intercepts/update_udev_hwdb

NOTE: ###### Generate rootfs #######
NOTE: Executing create_merged_usr_symlinks_rootfs ...
DEBUG: Executing shell function create_merged_usr_symlinks_rootfs
DEBUG: Shell function create_merged_usr_symlinks_rootfs finished
[... cut ...]
Note that if you provide several changes to a single variable
the last change will stay in effect.

See <https://qemu.org/contribute/report-a-bug> for how to report bugs.
More information on the QEMU project at <https://qemu.org>.

ERROR: The postinstall intercept hook 'update_font_cache' failed, details in /home/karleaves/Desktop/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/initramfs-ostree-image/1.0-r0/temp/log.do_rootfs
DEBUG: Python function do_rootfs finished
ERROR: Function failed: do_rootfs

```

Sadly it doesn't really give much info.

KarlEaves, 2019-07-18:
@MBetters I never found a real solution to this error... But I did find a work around to get it to build. Open meta/lib/oe/package-manager.py and change lines 454, 451, 461, to use bb.warn instead of bb.fatal. I hope that helps!

patrickvacek, 2020-07-30:
Looks like the original problem was resolved and the subsequent issues were worked around.

closed 2020-07-30 by patrickvacek