advancedtelematic/meta-updater #475: ExpansionError during parsing; (Also name error)

koruthu, 2019-02-08:
[docker.log](https://github.com/advancedtelematic/meta-updater/files/2846057/docker.log)

Not able to build with poky container, error log added.
bblayer.conf file script is as follows:

# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
 /workdir/poky/meta \
 /workdir/poky/meta-openembedded/meta-filesystems \
 /workdir/poky/meta-openembedded/meta-oe \
 /workdir/poky/meta-poky \
 /workdir/poky/meta-updater \
 /workdir/poky/meta-updater-raspberrypi \
 /workdir/poky/meta-yocto-bsp \
 "

I even tried "bitbake core-image-minimal" for complete build but same issue persists

OYTIS, 2019-02-11:
@koruthu 
I believe we've never tried to build it in docker. Mind sharing the Dockerfile?

koruthu, 2019-02-12:
> @koruthu
> I believe we've never tried to build it in docker. Mind sharing the Dockerfile?

koruthu, 2019-02-12:
@OYTIS please find the docker file, it was named "Dockerfile.bitbake"
[Dockerfile.txt](https://github.com/advancedtelematic/meta-updater/files/2854096/Dockerfile.txt)

OYTIS, 2019-02-14:
@koruthu OK, and the command line? Also it would be helpful to know what versions of the layers you're using. And local.conf of course.

Do you get this error only with meta-updater? That is if you exclude meta-updater and meta-updater-raspberrypi from bblayers.conf and all entries related to meta-updater from local.conf and just `bitbake postgresql`, will it succeed?

OYTIS, 2019-02-15:
@koruthu 
I've just accidentally reproduced this error by using `master` branch of `meta-openembedded` with `thud` branch of `poky`. Could it be your case as well?

koruthu, 2019-02-18:
I cannot find this error anymore when i used the command "bitbake core-image-sato".
The bulding started and went for many hours and ended with different error logs and 
[bblayers - Copy.txt](https://github.com/advancedtelematic/meta-updater/files/2873992/bblayers.-.Copy.txt)

The building was not complete out of about 6000 task it failed/stopped after 5500 task. (Sorry i couldn't provide the logs as its lost/overwritten)

patrickvacek, 2019-02-18:
@koruthu What platform are you trying to build for? It looks like you have not included a meta-updater platform compatibility layer. You probably want to add `BBLAYERS += " ${METADIR}/meta-updater-qemux86-64 "` or something similar to your `bblayers.conf`. Did you initialize your build directory with `meta-updater/scripts/envsetup.sh`? For basic cases, we initialize a repo based on https://github.com/advancedtelematic/updater-repo/.

koruthu, 2019-02-18:
Hello All,
Kindly let me know the details of the environment (OS version, other requirements etc.)you are using to run/develop this package.
We would like to replicate it and work this out, As other methods are not working for us. 

This is also related to issue thread: "How To run Aktualizer package in Eclipse IDE? #1069 " 

Our requirement is to build the images and run/use it in Qemu and then later on use it in Raspberry pie 3 device.

OYTIS, 2019-02-18:
@koruthu 

Our recommended "quick start" scenario is described in the README of [updater-repo](https://github.com/advancedtelematic/updater-repo). It supports, QEMU, Raspberry Pi3, and a couple of other platforms. You should be able to run it on any up-to-date Linux distribution. We regularly build it on Arch Linux, Ubuntu 16.04 and Ubuntu 18.04.

patrickvacek, 2020-07-30:
Closing due to lack of feedback.

closed 2020-07-30 by patrickvacek