advancedtelematic/meta-updater #482: Support for updating fpga bitstream

Bubu, 2019-02-15:
For our board (based on xilinx zynq) we need to load an fpga bitstream from u-boot before starting Linux.

Ideally this should just be handled alongside kernel and initrmfs which can be updated via ostree. But I couldn't find an obvious way to expand this to other files.

(Together with the bitstream the devivetree needs to be updated)

How should we go about solving this?

ricardosalveti, 2019-02-15:
Can the fpga bitstream be loaded as part of the fitimage (kernel+initrd+fpga)?

Bubu, 2019-02-15:
Good idea, I think we had this originally before switching to meta-updater layer.
Is fit image support something new here, I remember it not being supported...:thinking:.

OYTIS, 2019-02-18:
@Bubu FIT images have been added last March and have been greatly improved in August. It is mostly used/tested for Raspberry Pi, so you might want to take a look at `meta-updater-raspberrypi` layer as well.

Bubu, 2019-02-19:
Thanks, I'll take a look at fit images again. One question for clarification:

This would mean we'd put the bitstream and dtb file both into the fit image, right? And I guess we can also put an additional uboot script for updating the boot commands in there?

OYTIS, 2019-02-19:
> This would mean we'd put the bitstream and dtb file both into the fit image, right?

Yes, that's how FIT images work. We've never used this feature of U-Boot, but the code seems to be there. Just make sure that all relevant CONFIG_* options are enabled.

> And I guess we can also put an additional uboot script for updating the boot commands in there?

Actually you can, there is a dedicated image type in FIT for that ('script') and u-boot seems to know how to use it (with `source` command). Integrating that with Yocto/OE is a different story, I couldn't find any support for that in oe-core.

patrickvacek, 2020-07-30:
Closing due to lack of feedback.

closed 2020-07-30 by patrickvacek