advancedtelematic/meta-updater #483: dependency loop after thud update

Bubu, 2019-02-15:
When trying to update the meta-updater layer to `thud` bitbake immediately throws a dependency loop error.

```
Identifying dependency loops (this may take a short while)...

ERROR: 
Dependency loop #1 found:
  Task [projectdir]../external/meta-updater/recipes-core/images/initramfs-ostree-image.bb:do_image_complete (dependent Tasks ['dosfstools_4.1.bb:do_populate_sysroot', 'initramfs-ostree-image.bb:do_image', 'mtools_4.0.18.bb:do_populate_sysroot', 'u-boot-xlnx_2018.3.bb:do_populate_sysroot', 'parted_3.2.bb:do_populate_sysroot', 'initramfs-ostree-image.bb:do_image_cpio', 'zynq-bootfiles.bb:do_populate_sysroot'])
  Task [projectdir]../<board-layer>/recipes-kernel/linux/linux-yocto-stable_4.14.bb:do_bundle_initramfs (dependent Tasks ['initramfs-ostree-image.bb:do_image_complete', 'linux-yocto-stable_4.14.bb:do_install'])
  Task [projectdir]../<board-layer>/recipes-kernel/linux/linux-yocto-stable_4.14.bb:do_deploy (dependent Tasks ['depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'linux-yocto-stable_4.14.bb:do_populate_sysroot', 'linux-yocto-stable_4.14.bb:do_bundle_initramfs', 'linux-yocto-stable_4.14.bb:do_packagedata'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_fetch (dependent Tasks ['linux-yocto-stable_4.14.bb:do_deploy'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_prepare_recipe_sysroot (dependent Tasks ['linux-yocto-stable_4.14.bb:do_populate_sysroot', 'gcc-cross_8.2.bb:do_populate_sysroot', 'gcc-runtime_8.2.bb:do_populate_sysroot', 'glibc_2.28.bb:do_populate_sysroot', 'zynq-bootfiles.bb:do_fetch'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_configure (dependent Tasks ['zynq-bootfiles.bb:do_prepare_recipe_sysroot', 'zynq-bootfiles.bb:do_patch'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_compile (dependent Tasks ['zynq-bootfiles.bb:do_configure'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'zynq-bootfiles.bb:do_compile'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.31.bb:do_populate_sysroot', 'zynq-bootfiles.bb:do_install'])

Dependency loop #2 found:
  Task [projectdir]../external/meta-updater/recipes-core/images/initramfs-ostree-image.bb:do_image_complete (dependent Tasks ['dosfstools_4.1.bb:do_populate_sysroot', 'initramfs-ostree-image.bb:do_image', 'mtools_4.0.18.bb:do_populate_sysroot', 'u-boot-xlnx_2018.3.bb:do_populate_sysroot', 'parted_3.2.bb:do_populate_sysroot', 'initramfs-ostree-image.bb:do_image_cpio', 'zynq-bootfiles.bb:do_populate_sysroot'])
  Task [projectdir]../<board-layer>/recipes-kernel/linux/linux-yocto-stable_4.14.bb:do_bundle_initramfs (dependent Tasks ['initramfs-ostree-image.bb:do_image_complete', 'linux-yocto-stable_4.14.bb:do_install'])
  Task [projectdir]../<board-layer>/recipes-kernel/linux/linux-yocto-stable_4.14.bb:do_deploy (dependent Tasks ['depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'linux-yocto-stable_4.14.bb:do_populate_sysroot', 'linux-yocto-stable_4.14.bb:do_bundle_initramfs', 'linux-yocto-stable_4.14.bb:do_packagedata'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_fetch (dependent Tasks ['linux-yocto-stable_4.14.bb:do_deploy'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_unpack (dependent Tasks ['zynq-bootfiles.bb:do_fetch'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_patch (dependent Tasks ['zynq-bootfiles.bb:do_unpack', 'quilt-native_0.65.bb:do_populate_sysroot'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_configure (dependent Tasks ['zynq-bootfiles.bb:do_prepare_recipe_sysroot', 'zynq-bootfiles.bb:do_patch'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_compile (dependent Tasks ['zynq-bootfiles.bb:do_configure'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'zynq-bootfiles.bb:do_compile'])
  Task [projectdir]../<board-layer>/recipes-bsp/zynq-bootfiles/zynq-bootfiles.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.31.bb:do_populate_sysroot', 'zynq-bootfiles.bb:do_install'])
```

I'm not totally sure where the loop comes from but we have a task, `zynq-bootfiles` which needs the compiled dtb as an input, so it depends on `virtual/kernel:do_deploy`. And then we have `EXTRA_IMAGEDEPENDS_append_sota = " zynq-bootfiles"`

This has worked fine before the upgrade to thud.

patrickvacek, 2019-02-18:
I don't see any references to `zynq-bootfiles` in meta-updater or in fact any of the other layers we use in the updater-repo. Can you share your repo manifest?

OYTIS, 2019-02-18:
@Bubu 
Hi Marcus, great to hear from you again. It seems to me that the problem is `zynq-bootfiles` somehow landed in the dependencies of our initramfs image, `image-ostree-initramfs` where they do not belong. Would it be possible to `zynq-bootfiles` specifically to the dependencies of the main image?

OYTIS, 2019-02-18:
@Bubu That `zynq-bootfiles` depend on `linux-yocto-stable` also makes me suspicious. Is it really required?

Bubu, 2019-02-19:
@OYTIS I think all info should be in the initial post?

`zynq-bootfiles` depends on `virtual/kernel:do_deploy` because one input of zynq-bootfiles currently is the dtb file which gets generated in the kernel recipe.

And I'd guess this line: `EXTRA_IMAGEDEPENDS_append_sota = " zynq-bootfiles"` also adds the recipe as a dependency to `image-ostree-initramfs` which I'm not whether tat;s intended behaviour.

I guess I could play around with removing the `EXTRA_IMAGEDEPENDS` line and see what happens.

patrickvacek, 2019-02-19:
> And I'd guess this line: `EXTRA_IMAGEDEPENDS_append_sota = " zynq-bootfiles"` also adds the recipe as a dependency to `image-ostree-initramfs` which I'm not whether tat;s intended behaviour.

I was curious where that line came from, actually. It's not in any of our repos.

Bubu, 2019-02-19:
@patrickvacek It's in out board specific sota config:
```
$ cat classes/sota_zynq7.bbclass
OSTREE_BOOTLOADER ?= "u-boot"
EXTRA_IMAGEDEPENDS_append_sota = " zynq-bootfiles"
IMAGE_BOOT_FILES_sota += "zynq-bootfiles/*"
```

I can't really tell you *why* that's there, only that it was added on initial meta-updater integration.

OYTIS, 2019-02-19:
@Bubu @patrickvacek 

I believe `EXTRA_IMAGEDEPENDS` is still used for some machines like RCar boards for instance. I think when we started the integration, the dependency on linux kernel was not yet there, so it worked just fine.

You might try to replace EXTRA_IMAGEDEPENDS with what `meta-raspberrypi` uses:

```
do_image_wic[depends] += " zynq-bootfiles:do_deploy"
```

I _suspect_ it might solve your problem.

OYTIS, 2019-02-20:
@Bubu The reason why I find dependency of `zynq-bootfiles` on the kernel suspicious is that both kernel and dtb blobs are updatable (you need to add support for FIT images for dtb blobs, please again refer to the [boot script of RPi](https://github.com/advancedtelematic/meta-updater-raspberrypi/blob/master/recipes-bsp/u-boot-otascript/u-boot-otascript/uEnv.txt) as the most advanced one to the point) and bootfiles normaly are not, so a system update might render boot files outdated with unpredictable result.

Bubu, 2019-02-22:
@OYTIS your suggestion fixes this immediate problem. We'll look into switching to a fit image.

closed 2019-02-22 by Bubu