advancedtelematic/meta-updater #484: updating uboot from uboot

Bubu, 2019-02-15:
Some background:

In Xilinx's world the inital bootloader and uboot are provided by an image called `boot.bin` which is then loaded from a qspi flash. Given a boot.bin file, it's easy to update the qspi flash from uboot.

How would we update this file (and a metadata version file) via aktualizr/ostree? The problem we are struggling with is that uboot doen't have access to the deployed ostree repo and only things inside of this can be updated via OTA.

cajun-rat, 2019-02-16:
Hi,

One solution to this is to make U-Boot a secondary ECU. There is already support for multiple different types of secondaries, and there is infrastructure in Aktualizr to manage all the Uptane parts of being a secondary. I have a fork that is an example how to do this with docker-compose. This might be a good template: rather than calling docker-compose, instead give the update to Xilinx’s qspi flashing tool. 

https://github.com/advancedtelematic/aktualizr/pull/1046/files

Phil

Sent with care and attention from my mobile device

OYTIS, 2019-02-18:
Hi @Bubu,

we are considering adding support for updating the bootloader, but there are some difficulties associated with it. One of them being that the process of updating the bootloader is platform-dependent, and the more important one is that it's impossible to do it in a safe, atomic and reversible way.

One way to solve this problem for your project is to use the method proposed by @cajun-rat . You will need to fork aktualizr and add a new secondary pretty much like what's implemented in Phil's PR. Notice that our back end doesn't support adding secondaries to devices that have already been registered, so this will only enable bootloader update for new devices.

Also to access OSTree deployment from u-boot, you might check our recent changes to [u-boot script for Raspberry Pi](https://github.com/advancedtelematic/meta-updater-raspberrypi/blob/master/recipes-bsp/u-boot-otascript/u-boot-otascript/uEnv.txt), specifically the `bootcmd_getroot` and `bootcmd_fitconf` commands. You might implement something similar without secondaries: deploy the qspi bitsteam to the OSTree file system together with versioning information and save the current version to the permanent u-boot environment (don't forget the `saveenv`!).

patrickvacek, 2019-02-20:
@Bubu Unfortunately we don't currently have support for a concept of multiple updateable components on a single ECU. However, this can be faked as Phil has suggested: treat this u-boot file as a secondary and extend our existing secondary support as necessary for your specific case. Down the line, we may handle this more elegantly in tandem with the web UI.

patrickvacek, 2020-07-30:
Closing due to lack of feedback.

closed 2020-07-30 by patrickvacek