advancedtelematic/meta-updater #543: Unable to run init

alistair23, 2019-07-17:
I am trying to use OSTree with OE with RISC-V. I have ported meta-updater to a new board (freedom-u540). I have U-Boot starting where I then run these commands:
```
setenv bootargs earlycon=sbi earlycon=sbi console=ttySIF0 console=tty0 root=/dev/ram0 rw rootfstype=ext4 rootwait rootdelay=2 ramdisk_size=16384 ostree_root=/dev/mmcblk0p3
setenv devtype mmc
load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} uImage;
load ${devtype} ${devnum}:${distro_bootpart} ${ramdisk_addr_r} initramfs-ostree-image-freedom-u540.cpio.gz.u-boot
bootm ${kernel_addr_r} ${ramdisk_addr_r} $fdt_addr_r
```
Where `/dev/mmcblk0p3` contains my normal working rootFS, while `initramfs-ostree-image-freedom-u540.cpio.gz.u-boot` is the initrd produced by OE.

I see this failure during the kernel boot:

> [    8.067928] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
> [    8.078098] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
> [    8.085028] CPU: 0 PID: 1 Comm: init Not tainted 5.2.1 #1
> [    8.090405] Call Trace:
> [    8.092853] [<ffffffe0000394a8>] walk_stackframe+0x0/0xa6
> [    8.098223] [<ffffffe00003960e>] show_stack+0x2a/0x34
> [    8.103267] [<ffffffe0006871bc>] dump_stack+0x66/0x82
> [    8.108296] [<ffffffe00003dfa8>] panic+0xdc/0x246
> [    8.112983] [<ffffffe00003fd28>] do_exit+0x740/0x7a6
> [    8.117931] [<ffffffe0000407d0>] do_group_exit+0x2a/0xba
> [    8.123228] [<ffffffe000040878>] __wake_up_parent+0x0/0x22
> [    8.128699] [<ffffffe0000383d6>] ret_from_syscall+0x0/0xe
> [    8.134079] SMP: stopping secondary CPUs
> [    8.137994] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100 ]---

This is in my local.conf:
```
INHERIT_append_freedom-u540 = " sota"
DISTRO_FEATURES_append_freedom-u540 = " usrmerge"
DISTRO_FEATURES_NATIVE_append_freedom-u540 = " sota"
```

Any ideas why the init is dying? I'm using systemd as the init.

eu-smirnov, 2019-07-18:
Hi @alistair23, it's hard to say what exactly went wrong, but your bootargs look incomplete to me. Could you also post the kernel log part with the `Kernel command line:` message (or maybe even attach the full log)? There should be an `ostree=` parameter taken from `/boot/loader/uEnv.txt`.
Also, when you say `/dev/mmcblk0p3 contains my normal working rootFS` you mean the filesystem from `*.otaimg` image file, right?

alistair23, 2019-07-18:
I'm working from home today so I don't have access to my dev board, I can include the log tomorrow.

I don't see an `ostree=` parameter in the [QEMU example](https://github.com/advancedtelematic/meta-updater-qemux86-64/blob/master/recipes-bsp/u-boot/files/0001-Set-up-environment-for-OSTree-integration.patch) what should it be?

`/dev/mmcblk0p3` contains the rootFS that I was using before adding meta-updater. That is the rootFS was regenerated after adding meta-updater, but it's just the `--source rootfs` argument in the [wic file](https://github.com/riscv/meta-riscv/blob/master/wic/freedom-u540-opensbi.wks).

eu-smirnov, 2019-07-18:
This parameter is loaded into bootargs on line 45 in the QEMU example from `/boot/loader/uEnv.txt` file, which redefines the bootargs variable. Then, other parameters are appended on line 35.

alistair23, 2019-07-19:
Ah, I missed that. Where does `/boot/loader/uEnv.txt` come from? I see one in [this PR](https://github.com/advancedtelematic/meta-updater/pull/491/files). So if I take that file and load it it should work (obviously with the addresses updated)?

eu-smirnov, 2019-07-19:
This file should already exist in your rootfs. It's created [here](https://github.com/advancedtelematic/meta-updater/blob/d965046f12f9a29b166949b8c27bdf96fba6d3b7/classes/image_types_ota.bbclass#L57) and populated by ostree during the [deploy](https://github.com/advancedtelematic/meta-updater/blob/d965046f12f9a29b166949b8c27bdf96fba6d3b7/classes/image_types_ota.bbclass#L70) step. It will be also modified by ostree during installations.

alistair23, 2019-07-19:
Ah, that needs `sota` enabled in `DISTRO_FEATURES` which I don't have as I see this error:

> ERROR: Task do_image_wic in /scratch/alistair/yocto/oe-master/meta/recipes-core/images/core-image-minimal.bb depends upon non-existent task IMAGE_CMD_ota-ext4 in /scratch/alistair/yocto/oe-master/meta/recipes-core/images/core-image-minimal.bb
> ERROR: Command execution failed: 1

from this line:

```
do_image_wic[depends] += "${@bb.utils.contains('DISTRO_FEATURES', 'sota', '%s:do_image_ota_ext4' % d.getVar('IMAGE_BASENAME'), '', d)}"
```

ricardosalveti, 2019-07-19:
You might also require a specific wic file after you get the base image to work, to use the otaimg image instead.

I got it to work with meta-lmp, which is a distro we maintain that also uses meta-updater, take a look at https://github.com/ricardosalveti/meta-lmp/commit/24cd590c88c5072366857b38b300c794d1ec3c8d for the changes I did. Also recorded https://asciinema.org/a/258203 that shows it booting with opensbi, u-boot, fitimage and ostree.

eu-smirnov, 2019-07-22:
@alistair23 Do you also have `INHERIT += " sota"` in your local.conf?

alistair23, 2019-07-22:
I do have `INHERIT += " sota"`

alistair23, 2019-07-22:
> I got it to work with meta-lmp, which is a distro we maintain that also uses meta-updater, take a look at [ricardosalveti/meta-lmp@24cd590](https://github.com/ricardosalveti/meta-lmp/commit/24cd590c88c5072366857b38b300c794d1ec3c8d) for the changes I did. Also recorded https://asciinema.org/a/258203 that shows it booting with opensbi, u-boot, fitimage and ostree.

That's awesome! That is what I'm trying to get working. Do you have any interest in sending this support to meta-riscv?

alistair23, 2019-07-22:
This is the changes I have to meta-updater and no other changes: https://github.com/alistair23/meta-updater/tree/alistair/freedom-u540

ricardosalveti, 2019-07-23:
> That's awesome! That is what I'm trying to get working. Do you have any interest in sending this support to meta-riscv?

Sure, most of it is distro specific, but I can send the more generic pieces, will try to get to this tomorrow.

alistair23, 2019-07-23:
That would be great! Looking at your repo you should no longer need a specific wik file as meta-riscv has one for you.

alistair23, 2019-07-23:
Got it building!

I need to remove the wic image types (sota adds its own after the ota-ext4 image). So this line does it:
```
IMAGE_FSTYPES_remove = "wic.gz"
```

alistair23, 2019-07-23:
I'm now seeing this:

> [    0.000000] Kernel command line: earlycon=sbi console=ttySIF0 console=ttyS0 ostree_root=/dev/mmcblk0p3 root=LABEL=otaroot rootfstype=ext4 ostree=/ostree/boot.1/poky/4a8ff592e1ca7d03334f3c1e32b011736b24f6519633383b36619b470f7ee5ca/0
> ...
> [    8.037110] mmc0: host does not support reading read-only switch, assuming write-enable
> [    8.049877] Run /init as init process
> [    8.073339] mmc0: new SDXC card on SPI
> /init[1]: Starting OSTree initrd script
> [    8.083352] mmcblk0: mmc0:0000 SP128 119 GiB
> /init[1]: mounting FS: proc /proc
> /init[1]: mounting FS: sysfs /sys
> /init[1]: mounting FS: devtmpfs /dev
> /init[1]: mounting FS: devpts /dev/pts
> /init[1]: mounting FS: tmpfs /dev/shm
> /init[1]: mounting FS: tmpfs /run
> [    8.120177] GPT:Primary header thinks Alt. header is not at the end of the disk.
> [    8.134289] GPT:13975585 != 249737215
> [    8.141655] GPT:Alternate GPT header not at the end of the disk.
> [    8.153719] GPT:13975585 != 249737215
> [    8.161094] GPT: Use GNU Parted to correct GPT errors.
> [    8.171483]  mmcblk0: p1 p2 p3
> ...
> [   13.252477] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
> ostree-prepare-root: Couldn't find specified OSTree root '/sysroot//ostree/boot.1/poky/4a8ff592e1ca7d03334f3c1e32b011736b24f6519633383b36619b470f7ee5ca/0': No such file or directory
> [   13.290987] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
> [   13.306225] CPU: 3 PID: 1 Comm: init Not tainted 5.2.2 #1
> [   13.317070] Call Trace:
> [   13.322037] [<ffffffe0000394a8>] walk_stackframe+0x0/0xa6
> [   13.332877] [<ffffffe00003960e>] show_stack+0x2a/0x34
> [   13.343043] [<ffffffe00068729c>] dump_stack+0x66/0x82
> [   13.353194] [<ffffffe00003dfa8>] panic+0xdc/0x246
> [   13.362656] [<ffffffe00003fd28>] do_exit+0x740/0x7a6
> [   13.372640] [<ffffffe0000407d0>] do_group_exit+0x2a/0xba
> [   13.383320] [<ffffffe000040878>] __wake_up_parent+0x0/0x22
> [   13.394347] [<ffffffe0000383d6>] ret_from_syscall+0x0/0xe
> [   13.405197] SMP: stopping secondary CPUs
> [   13.413107] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100 ]---

I'm assuming `/ostree/boot.1/poky/4a8ff592e1ca7d03334f3c1e32b011736b24f6519633383b36619b470f7ee5ca/0` is the problem, where does that come from?

ricardosalveti, 2019-07-24:
I would have to confirm, but I believe the issue here is the standard wic file you are using. The one I created was needed because of the --source otaimage parameter (which is needed because the standard rootfs file is not the actual ostree rootfs, which is named ota-ext4).

I started separating the minimal to get the build going, and you can give a try by running:
`repo init -u git://github.com/ricardosalveti/meta-riscv -b updater -m tools/manifests/riscv-yocto-updater.xml`.

This was enough for me to get the core-image-full-cmdline working with u-boot, fitimage and ostree.

Will revisit the changes tomorrow and propose to the upstream branches in a more generic way.

ricardosalveti, 2019-07-24:
My changes can also be found at:
 - https://github.com/ricardosalveti/meta-riscv/tree/updater
 - https://github.com/ricardosalveti/meta-updater-riscv

patrickvacek, 2019-07-24:
@ricardosalveti That's awesome! By coincidence I'm updating the list of supported boards/platforms in the meta-updater README. I'd like to mention your meta-updater-riscv layer there; would you be okay with that?

ricardosalveti, 2019-07-24:
@patrickvacek sure, also sending another pr to add freedom support at meta-updater itself.

ricardosalveti, 2019-07-24:
Created https://github.com/advancedtelematic/meta-updater/pull/565 to add the machine specific configuration for it.

alistair23, 2019-07-24:
> --source otaimage parameter (which is needed because the standard rootfs file is not the actual ostree rootfs, which is named ota-ext4).

Yep! That is it. I have it working now. I'll try now with your PR and a fitImage.

alistair23, 2019-07-24:
Going to close this PR as it's working for me and @ricardosalveti is currently sending PRs

closed 2019-07-24 by alistair23