advancedtelematic/meta-updater #555: Recent commit breaking compatibility with thud and warrior

albard00, 2019-07-19:
Hi there,
I just discovered a small issue that appear recently:

https://github.com/advancedtelematic/meta-updater/commit/12dbe3116ed539767ca5d283dd558b565e864a2c is breaking the compatibility with warrior and thud because openssl_1.1.1c.bb does not exist. The openssl version in thud and warrior is 1.1.1b. 1.1.1c is only present in master

Thanks for your work

Best regards

lbonn, 2019-07-19:
Hello and thanks for the report.

We currently don't recommend using the master branch of meta-updater with other yocto release branches as we backport changes that make sense to the corresponding meta-updater branches.

You can use our thud branch, but we indeed don't have a warrior branch at the moment, that's on our list.

patrickvacek, 2019-07-19:
@albard00 Yes, it's probably time for a warrior branch. Until that commit, master was more-or-less compatible with warrior, but that unfortunately isn't true anymore. Thanks for letting us know!

lbonn, 2019-07-19:
I've proposed to upstream the patch to openembedded. If accepted, we will be able to drop it here.

lbonn, 2019-07-22:
Reference: https://patchwork.openembedded.org/patch/163219/. When merged it will actually break our master because of the duplicate patch :). Then we'll just drop it and this issue will be solved.

lbonn, 2019-07-23:
Fixed by #559

closed 2019-07-23 by lbonn