advancedtelematic/meta-updater #564: Missing warrior branch

agners, 2019-07-24:
Is there a branch for warrior planned? We currently use `550ef5c1c32be58cca91f80afbedc23d5e9b79b6` for warrior, but I am not sure up to which point I can safely fast-forward...

patrickvacek, 2019-07-24:
Yes, we are planning it. Until recently, master worked fine with warrior, and I think when https://github.com/advancedtelematic/meta-updater/issues/555 was resolved that means that warrior will work again with master, but I haven't tested it myself. In any case, we will make an official branch soon.

agners, 2019-07-24:
Oh I see this has been discussed before. I am sorry, I could have looked through issues a bit more before creating a new issue.

Ok thanks for the information.

patrickvacek, 2019-11-14:
Closing; this was done a while ago and now it's actually time for zeus, which we will be adding soon.

closed 2019-11-14 by patrickvacek