advancedtelematic/meta-updater #603: sstate signature issues

shr-project, 2019-09-12:
There are quite a few issues reported by oe-core/scripts/sstate-diff-machines.sh script in meta-updater recipes, mostly allarch recipes depending on TUNE_PKGARCH without setting SIGGEN_EXCLUDERECIPES_ABISAFE/SIGGEN_EXCLUDE_SAFE_RECIPE_DEPS

ERROR: 32 issues were found in these recipes: aktualizr-device-prov aktualizr-device-prov-creds aktualizr-device-prov-hsm aktualizr-uboot-env-rollback networkd-dhcp-conf primary-network-config secondary-network-config

Can you please run the script and fix the issues it reports?

patrickvacek, 2019-09-13:
Hello @shr-project, thanks for reporting this issue! I'm attempting to look into it, but unfortunately haven't been able to wrap my head around the tooling yet. I ran `../poky/scripts/sstate-diff-machines.sh --tmpdir=tmp --targets=core-image-minimal` (with and without `--analyze`), and then I diffed the list.M files as described [here](https://lists.yoctoproject.org/pipermail/poky/2014-August/009840.html). However, I can't find the sigdata files referenced in those. Thus I can't get `bitbake-diffsigs` to provide meaningful output. What do I do next?

Is there documentation for how to use these tools?

shr-project, 2019-09-13:
Hi @patrickvacek,
there is short documentation in the script itself.

You need to test it on something which includes all the recipes from meta-updater (e.g. world, core-image-minimal won't include them).

--analyze can be used only when you provide meaningful --machines parameter e.g. qemux86 qemux86copy qemuarm

Here is the output from:
oe-core/scripts/sstate-diff-machines.sh --tmpdir=tmp-glibc --machines="qemux86 qemux86copy qemuarm" --targets=world --analyze
with meta-updater layer included.
For qemux86copy you'll need meta-selftest layer (./meta-selftest/conf/machine/qemux86copy.conf) or you can cherry-pick this commit to oe-core:
http://git.openembedded.org/openembedded-core-contrib/commit/?h=jansa/master&id=dd27e838ad87dcc82e79c1284c76304d4128a230

http://dpaste.com/22CF7JD

Simple exemple would be recipes-sota/aktualizr/aktualizr-device-prov.bb which inherits allarch, but has
`
DEPENDS = "aktualizr aktualizr-native openssl-native"
RDEPENDS_${PN}_append = "${@' aktualizr-device-prov-creds' if d.getVar('SOTA_DEPLOY_CREDENTIALS') == '1' else ''}"
`

It cannot depend on TUNE_PKGARCH aktualizr and while being allarch (unless you mark this dependency as ABI safe with SIGGEN_EXCLUDERECIPES_ABISAFE/SIGGEN_EXCLUDE_SAFE_RECIPE_DEPS) neither of these 3 build time dependencies, doesn't seem to be needed in this case anyway.

Similarly aktualizr-device-prov-creds.bb probably doesn't need build time dependency on target aktualizr and so on.

patrickvacek, 2019-10-07:
Hello @shr-project, thanks for the tips. However, I'm not getting any output when I run the script as you suggest. I don't see anything in the .sigdata or list.M files. What might I be doing wrong?

Also, that link you pasted to dpaste.com expired. I must've waited too long to look into this. Hopefully I can figure out how to get the output myself, but if not, that would be helpful to see.

patrickvacek, 2019-10-08:
I played around some more today but did not get any meaningful results.

I tried running `../poky/scripts/sstate-diff-machines.sh --tmpdir=tmp --machines="qemux86 qemux86-64 qemuarm" --targets=world --analyze` since `qemux86-64` is an actual machine type that we explicitly support. That yielded results, but most of them were in `do_populate_sysroot` steps for recipes that we don't actually manipulate, like `android-tools-conf`. There are also some for `do_configure`, including `u-boot-fw-utils`, which is one of our recipes, but I can't figure out how to understand what went wrong.

I tried using `bitbake-diffsigs` as suggested in the script's comments, but I cannot find a directory named `stamps.<timestamp>`. How do I generate or find that directory? Or how else do I provide input to `bitbake-diffsigs`?

I also tried running `../poky/scripts/sstate-diff-machines.sh --tmpdir=tmp --machines="raspberrypi3 raspberrypi3-64 qemux86-64" --targets=world --analyze`, i.e. three machines that we explicitly support, and I got no output.

What might you be doing differently to get meaningful output? I'm afraid I'm stuck at this point as to how to proceed.

shr-project, 2019-10-08:
See the --analyze description
`  --analyze
        Show the differences between MACHINEs. It assumes:

        * First 2 MACHINEs in --machines parameter have the same TUNE_PKGARCH

        * Third optional MACHINE has different TUNE_PKGARCH - only native and allarch recipes are compared).

        * Next MACHINEs are ignored
`

qemux86 and qemux86-64 have different architecture so the script is correctly showing that all TUNE_PKGARCH recipes have different signatures. Use the MACHINEs I've mentioned before or e.g. raspberrypi3 and raspberrypi4 as long as you use the same DEFAULTTUNE for them.

The stamps.<timestamp> directory should be created inside tmp directory (--tmpdir param).

shr-project, 2019-10-10:
I've re-uploaded output from
oe-core/scripts/sstate-diff-machines.sh --tmpdir=tmp-glibc --machines="qemux86 qemux86copy qemuarm" --targets=world --analyz
again, see:
http://paste.ubuntu.com/p/GxPYgxf4sY/

shr-project, 2019-10-11:
First issue is that your environment doesn't respect MACHINE variable set from shell environment, check if you have MACHINE included in BB_ENV_EXTRAWHITE.

As you can see in your output all the bitbake calls had:
`MACHINE              = "qemux86-64"`
instead of qemux86, qemux86copy, qemux86-64

And then
`ERROR: no sigdata files were generated for MACHINE qemux86 in /home/patrick/Code/updater_qemu_master/build/tmp-glibc/stamps`
because it genearted stamps for qemux86-64 set in your environment instead of qemux86 set in shell environment by the sstate-diff-machines.sh

> If I change the tmpdir to my normal bitbake tmpdir, I don't get those complaints, but I also don't get anything interesting. 

I don't know what other tmpdir you were using before, but you need to use the normal bitbake one, see the `scripts/sstate-diff-machines.sh --help`:

>  --tmpdir=<tmpdir>
>        Specify tmpdir, will use the environment variable TMPDIR if it is not specified.
>        Something like /OE/oe-core/tmp-eglibc (no / at the end).

In your case use /home/patrick/Code/updater_qemu_master/build/tmp-glibc (it can be relative to workdir, so --tmpdir=tmp-glibc should work fine.

Not surprising that the output isn't interesting when it has nothing to compare (all 3 runs used qemux86-64 MACHINE) fix that and you should get similar output to what I've shared before.

patrickvacek, 2019-10-11:
> First issue is that your environment doesn't respect MACHINE variable set from shell environment, check if you have MACHINE included in BB_ENV_EXTRAWHITE.

Thanks, good catch! Yes, by default we specify `MACHINE` in `local.conf`, and when I removed that, it worked much more closely to what you indicated.

However, I still don't see the `stamps.` directories, so it's hard for me to dig very deep here. How are those supposed to get created?

I looked around the recipes that were complained about, and I tried to read up on the variables you mentioned, but was only able to find a few things that looked obviously wrong to me. I've made a PR (https://github.com/advancedtelematic/meta-updater/pull/612) with those fixes. However, re-running the script still yields a number of complaints (albeit a few less than before).

I'm guessing that one of the problems is that we have allarch recipes that depend on the aktualizr recipe, which is not allarch. However, in most cases, the allarch recipe (`recipes-sota/aktualizr/aktualizr-device-prov.bb` for example) is merely taking a config file provided by the aktualizr recipe and installing it in the image. How do I express that dependency? It certainly doesn't depend on the architecture that aktualizr was built for, but the descriptions of `SIGGEN_EXCLUDE_SAFE_RECIPE_DEPS` and `SIGGEN_EXCLUDERECIPES_ABISAFE` don't sound right here. ("recipes that are completely stable and will never change" is particularly intimidating: what does "never change" really mean here?) Have we misunderstood when to use allarch?

At any rate, that doesn't seem to be the underlying issue for all the recipes; some of them do not have any stated dependencies. Unfortunately, I'm at a loss to understand what is wrong with those.

shr-project, 2019-10-11:
> However, I still don't see the stamps. directories, so it's hard for me to dig very deep here.

These are created by bitbake in directory specified by TMPDIR/STAMPS_DIR variable:
`meta/conf/bitbake.conf:STAMPS_DIR ?= "${TMPDIR}/stamps"`

The script uses stamp files from this directory:
`${tmpdir}/stamps/`
and creates a copy of it in the directory:
`OUTPUT=${tmpdir}/sstate-diff/`date "+%s"``
where tmpdir is what you specified in --tmpdir when calling the sstate-diff-machines.sh script.

> I'm guessing that one of the problems is that we have allarch recipes that depend on the aktualizr recipe, which is not allarch.
Yes that's the issue for bunch of them as mentioned in 3rd comment here.

> Have we misunderstood when to use allarch?

Yes, with allarch all MACHINEs should be using the same sstate archive and sstate signature (as checked by this script). Depending on TUNE_PKGARCH recipe from allarch recipe causes the created packages to be in "all" package feed, but still rebuilt every time you change the MACHINE you're building for - making whole allarch inherit very inefficient (because it's rebuilt or recreated from sstate every single time for multi MACHINE builds and also the "alternating" sstate archives might be removed e.g. by scripts/sstate-cache-management.sh which leaves only one latest version of each sstate archive). Using allarch inherit in this case doesn't buy you anything only makes it worse.

If you use `SIGGEN_EXCLUDE_SAFE_RECIPE_DEPS` or `SIGGEN_EXCLUDERECIPES_ABISAFE` then you can get rid of these unnecessary rebuilds (as long as the config file you package here is exactly the same for all MACHINEs), but you also need to ensure that the version of e.g. aktualizr-shared-prov is changed whenever the config file in aktualizr is modified.

It looks like you're installing the config file from ${STAGING_DIR_NATIVE}${libdir}/sota which is good in this case, because there is only one native sysroot, so the config file is already the same for all MACHINEs.

> Unfortunately, I'm at a loss to understand what is wrong with those.

Once you have the sstate-diff directory correctly populated you need to use bitbake-diffsigs on the files there to see what difference caused it, it might take couple steps, because most likely it will just show that some other recipe changed the signature, so you need to traverse the dependency tree a bit to find out the root cause - for aktualizr it's clear when it has different PACKAGE_ARCH.

patrickvacek, 2019-10-14:
> Yes, with allarch all MACHINEs should be using the same sstate archive and sstate signature (as checked by this script). Depending on TUNE_PKGARCH recipe from allarch recipe causes the created [...]

I appreciate the thorough explanation! I think I've fixed that part up now. I've updated my [PR](https://github.com/advancedtelematic/meta-updater/pull/612) with some new changes.

> The script uses stamp files from this directory:
> `${tmpdir}/stamps/`
> and creates a copy of it in the directory:
> `OUTPUT=${tmpdir}/sstate-diff/`date "+%s"``

I think I get it now. I was expecting there to be a directory literally named `stamps.<timestamp>`. I didn't realize the files I needed were already there, just in a differently-named directory. The first thing I've tried is this:

```
$ bitbake-diffsigs tmp/sstate-diff/1571038352/qemux86*/core2-32-poky-linux/aktualizr/1.0+gitAUTOINC+d13ff1ceec-7.do_install.sigdata*
NOTE: Starting bitbake server...
basehash changed from e200ad0a6862103734cb5ddd30dedb3c7047b98fcd78ac5f1f07688ac9e2b7e6 to c5c114d92f1cb7ce42fd5a0f049167dca0b95f1847df4224868c7fb3085b5c71
Variable MACHINE value changed from 'qemux86copy' to 'qemux86'
```

Is it a problem that the recipe depends on the value of `MACHINE`? We use that variable as a default value to populate `SOTA_HARDWARE_ID`, which we write to a file on the image. This also affects aktualizr-native, and if I'm understanding correctly, this is also why I see this:

```
$ bitbake-diffsigs tmp/sstate-diff/1571044038/qemux86{,copy}/*/aktualizr-device-prov-creds/*.do_prepare_recipe_sysroot.sigdata.*
NOTE: Starting bitbake server...
Hash for dependent task aktualizr/aktualizr_git.bb:do_populate_sysroot:virtual:native:/home/patrick/Code/updater_qemu_master/meta-updater/recipes-sota/aktualizr/aktualizr_git.bb changed from 6cbab430ab82001707e7a9ab62adbf9022121f13b72ba598f45451be11358298 to d9d4bd4ecb7efbac165b091b3724f74d16de7eb4c7b2736e5332ca303f6d8fc0
Unable to find matching sigdata for virtual:native:/home/patrick/Code/updater_qemu_master/meta-updater/recipes-sota/aktualizr/aktualizr_git.bb:do_populate_sysroot with hashes 6cbab430ab82001707e7a9ab62adbf9022121f13b72ba598f45451be11358298 or d9d4bd4ecb7efbac165b091b3724f74d16de7eb4c7b2736e5332ca303f6d8fc0
```

This `MACHINE` issue looks to be the root cause of all the remaining complaints. I'm not sure how to address that.

patrickvacek, 2019-10-17:
All issues resolved as far as I can see in https://github.com/advancedtelematic/meta-updater/pull/612. Thanks for all the help with this!

patrickvacek, 2019-10-18:
Closed by https://github.com/advancedtelematic/meta-updater/pull/612.

closed 2019-10-18 by patrickvacek

shr-project, 2019-10-25:
There is one more issue in networkd-dhcp-conf:
first it inherits allarch:
https://github.com/advancedtelematic/meta-updater/blob/master/recipes-connectivity/networkd-dhcp-conf/networkd-dhcp-conf.bb#L7
then it RDEPENDS on TUNE_PKGARCH systemd:
https://github.com/advancedtelematic/meta-updater/blob/master/recipes-connectivity/networkd-dhcp-conf/networkd-dhcp-conf.bb#L18
and then it sets PACKAGE_ARCH to MACHINE_ARCH:
https://github.com/advancedtelematic/meta-updater/blob/master/recipes-connectivity/networkd-dhcp-conf/networkd-dhcp-conf.bb#L23

It was probably shown in your sstate-diff-machines.sh output as well, see:
` === Comparing signatures for task do_package_write_ipk.sigdata between qemux86 and qemux86copy ===`
`ERROR: networkd-dhcp-conf different signature for task do_package_write_ipk.sigdata between qemux86 and qemux86copy`
`Hash for dependent task systemd/systemd_243.bb:do_packagedata changed from ef01375cec5695f9b636163f7d5c1c03beb4428da3c47ff4d0cefcc70d3f25a3 to 29f2f04cb4c15515831e15ecaeffc4c72970f1eec5ad88e8fb4206f111352284`

Easiest fix is to just drop allarch (setting PACKAGE_ARCH before inherit allarch would also help, but dropping it is better).

patrickvacek, 2019-10-28:
> There is one more issue in networkd-dhcp-conf:

Interesting, I don't see that error when I run the script. Not sure why it would be different. But anyway, I agree, setting `allarch` doesn't seem to be helpful here if we end up setting `PACKAGE_ARCH` to `MACHINE_ARCH`. I'll make a PR.

patrickvacek, 2019-10-28:
See https://github.com/advancedtelematic/meta-updater/pull/622.