advancedtelematic/meta-updater #610: Error building meta-updater in Yocto for raspberrypi

GWLindberg, 2019-09-18:
I'm trying to add ostree to a Yocto THUD raspberry pi build. I've added the layers to the bblayers.conf file:
  ${BSPDIR}/sources/meta-openembedded/meta-filesystems \
  ${BSPDIR}/sources/meta-updater \                 
  ${BSPDIR}/sources/meta-updater-raspberrypi \     

And enabled 'sota' and some configuration in the local.conf file:

# make sure to inherit OTA functionality from the meta-updater* layers
INHERIT += " sota"    
DISTRO_FEATURES += " usrmerge"

When I try to build I get the following error:

ERROR: ostree-v2018.9-r0 do_unpack: gitsm: submodule unpack failed: UnpackError Unpack failure for URL: 'gitsm://gitlab.gnome.org/GNOME/libglnx.git;protocol=https;name=libglnx;subpath=libglnx;bareclone=1;nobranch=1'. No up to date source found: clone directory not available or not up to date: /home/gwilson/Qt/Qt-5.13.1/Yocto-build-RPi3/build-raspberrypi3/../downloads/git2/gitlab.gnome.org.GNOME.libglnx.git; shallow clone not enabled
ERROR: ostree-v2018.9-r0 do_unpack: Unpack failure for URL: 'gitsm://gitlab.gnome.org/GNOME/libglnx.git;protocol=https;name=libglnx;subpath=libglnx;bareclone=1;nobranch=1'. No up to date source found: clone directory not available or not up to date: /home/gwilson/Qt/Qt-5.13.1/Yocto-build-RPi3/build-raspberrypi3/../downloads/git2/gitlab.gnome.org.GNOME.libglnx.git; shallow clone not enabled

Am I missing some configuration?

lbonn, 2019-09-20:
That's odd indeed! Could you maybe describe briefly which layer revisions you are using (in particular, poky), so that we can have a better shot at reproducing it?

And the two lines at the top are the only ones you've added to your configuration?

GWLindberg, 2019-09-20:
Hi Ibonn,
I'm using Poky Distribution 2.6.2. I'm actually building on a base of Qt's boot2qt system, but it is almost entirely Yocto, just some extra Qt build stuff.

For the initramfs i am using 'initramfs-ostree-image' because I couldn't find a better one in Yocto.

I'll attach both my bblayers. conf & local.conf files so that you can see all of what I have in them.
[conf-files.zip](https://github.com/advancedtelematic/meta-updater/files/3636526/conf-files.zip)

My image recipe is based on Qt's boot2qt image with added local recipes, but it has nothing in it about ostree. if you need to see it I can upload it also.

Thanks for taking a look,
Greg

GWLindberg, 2019-09-20:
I should have said earlier that I had pulled the meta-updater* directories on August 28th if it makes any difference.

GWLindberg, 2019-09-20:
I just did a pull on meta-updater* and tried to rebuild, I got the same error.

lbonn, 2019-09-23:
There are a couple of bug fix backports on the gitsm fetcher between 2.6.2 and 2.6.3. From a quick glance, it doesn't look like the problem you're having but I would recommend using the last 2.6 release, which will have fixes for recent bugs and CVEs.

Unfortunately, I could not reproduce the issue directly but another thing you could try is `bitbake -c cleanall ostree` and `rm -rf /home/gwilson/Qt/Qt-5.13.1/Yocto-build-RPi3/build-raspberrypi3/../downloads/git2/gitlab.gnome.org.GNOME.libglnx.git*` before a rebuild.

patrickvacek, 2019-09-23:
This could indeed be something with the git fetching logic, since https://gitlab.gnome.org/GNOME/libglnx is directly clonable and we know there has been a history of submodule fetching problems. Odd, though, that we haven't seen this specific issue before.

patrickvacek, 2019-11-14:
Is this still a problem, or can we close this? This seems like it was either transient or related to changes in bitbake's git fetching logic, and we haven't been able to reproduce it. Also, OSTree in master is now in meta-openembedded.

GWLindberg, 2019-11-14:
Hi Patrick,
I haven't been able to work on this recently. If it has been put into meta-openembedded, I will (eventually) be able to upgrade my build system to that and will try again.

Thanks for looking into it anyway,

Greg

patrickvacek, 2019-11-15:
Closing for now, but feel free to reopen or open another if it emerges again.

closed 2019-11-15 by patrickvacek