advancedtelematic/meta-updater #684: Use native dependencies instead of HOSTTOOLS where possible

shr-project, 2020-02-15:
As commented in:
https://github.com/advancedtelematic/meta-updater/commit/d89f02d9eecbfe4df3bcc380b7d32ecf9f9ba933

Adding python to HOSTTOOLS is a bit toxic, because every recipe will now see python in PATH.

It would be better to use pythonnative only where really needed (and incrementally update to use python3 with python3native). Adding it to HOSTTOOLS just hides issues like this.

It also breaks gpsd from meta-oe, which wasn't detecting native python since python was removed from HOSTTOOLS in
http://git.openembedded.org/openembedded-core/commit/?id=5f8f16b17f66966ae91aeabc23e97de5ecd17447
and now it does detect it only in builds which include meta-updater with this commit:
`3a2fecf meta: move HOSTTOOLS definitions to sota.conf.inc`

resulting in:
```
--- 3.19-r0-fail/temp/log.do_compile    2020-02-15 09:08:14.234156815 -0800
+++ 3.19-r0-ok/temp/log.do_compile      2020-02-15 08:43:25.899427562 -0800
@@ -60,21 +60,17 @@
 Checking if compiler accepts -Wreturn-type... yes
 Checking if compiler accepts -Wstrict-prototypes... yes
 Checking if compiler accepts -Wvla... yes
-Checking whether python program exists.../jenkins/mjansa/build-webos-github-master/BUILD/hosttools/python
-Obtaining Python exe path... yes
-Obtaining Python lib dir... yes
-Obtaining Python config vars... ok
+Checking whether python program exists...no
+Target Python doesn't exist - disabling Python.
 Altered configuration variables:
 qt = False (default True): build QT bindings
 manbuild = False (default True): build help in man and HTML formats
 nostrip = True (default False): don't symbol-strip binaries at link time
+python = False (default True): build Python support and modules.
 prefix = /usr (default /usr/local): installation directory prefix
 sysroot = /jenkins/mjansa/build-webos-github-master/BUILD/work/raspberrypi4-webos-linux-gnueabi/gpsd/3.19-r0/recipe-sysroot (default ): cross-development system root
 libdir = /usr/lib (default lib): system libraries
-WARNING: Python module pycairo not found.
-    xgps and xgpsspeed will not be installed
-WARNING: Python module pygobject not found.
-    xgps and xgpsspeed will not be installed
+GPS regression tests suppressed because socket_export or python is off.
....
```

patrickvacek, 2020-02-17:
What is it that actually requires python2? We try to write our scripts to be python3-compliant, and if anything isn't, I would gladly take the time to fix it.

patrickvacek, 2020-07-30:
Closing due to lack of feedback. I'd be surprised if we really are depending on python2 for anything. Feel free to reopen if there is a specific problem we can address.

shr-project, 2020-07-30:
> What is it that actually requires python2?

That was discussed in the commit which introduced it:
https://github.com/advancedtelematic/meta-updater/commit/d89f02d9eecbfe4df3bcc380b7d32ecf9f9ba933

> Feel free to reopen if there is a specific problem we can address.

It would be better to remove the HOSTTOOLs additions from meta-updater:

classes/sota_qemux86-64.bbclass:HOSTTOOLS_NONFATAL += "x86_64-linux-gnu-gcc"
conf/distro/sota.conf.inc:HOSTTOOLS += "git sync sha256sum"
conf/distro/sota.conf.inc:HOSTTOOLS_NONFATAL += "java repo python"

and use native dependencies where needed.

I've unblocked our builds by removing it in webOS integration layer for meta-updater here:
https://github.com/shr-project/meta-webosose/commit/8ae2188d04b5dcccb305c26c73c226ae05ea1e77
but the issue exists for other meta-updater users.

closed 2020-08-03 by patrickvacek