advancedtelematic/meta-updater #698: Error building core-image-base

clem844, 2020-03-19:
I'm on zeus branch

When building "core-image base", this error occurs:
```
ERROR: core-image-base-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs --setopt=logdir=/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/temp --repofrompath=oe-repo,/home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo --nogpgcheck install aktualizr aktualizr-shared-prov fit-conf openssh os-release ostree packagegroup-base-extended packagegroup-core-boot psplash-raspberrypi run-postinsts systemd-journald-persistent virtual/network-configuration wifi-systemd-service locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.2.2
cachedir: /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Thu 19 Mar 2020 01:07:10 PM UTC.
Last metadata expiration check: 0:00:01 ago on Thu 19 Mar 2020 01:07:10 PM UTC.
No module defaults found
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: package packagegroup-base-extended-1.0-r83.raspberrypi3 requires packagegroup-base, but none of the providers can be installed
  - package packagegroup-base-1.0-r83.raspberrypi3 requires packagegroup-base-wifi, but none of the providers can be installed
  - conflicting requests
  - nothing provides wireless-regdb-static needed by packagegroup-base-wifi-1.0-r83.raspberrypi3
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

ERROR: Logfile of failure stored in: /home/tce/yocto_dist/build/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_rootfs.22978
ERROR: Task (/home/tce/yocto_dist/poky/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'
```

Building the same image with "poky" instead of "poky-sota-systemd" works with no errors. 
Also, I can build core-minimal-image without any problems with "poky-sota-systemd"

patrickvacek, 2020-03-19:
Interesting, it looks like the trouble is with `wireless-regdb-static`. I'm not sure how that's getting pulled in, but my guess is that it has something to do with `wifi-systemd-service` in our meta-updater-raspberrypi layer. Did you set `RPI_WIFI_ENABLE` to `1` by chance? If you set it to `0` does that help?

@mike-sul do you have other advice or ideas here?

And would you mind sharing your bblayers.conf (and possibly local.conf) with us? You can send us an email at otaconnect.support@here.com if you'd prefer that to posting it here; up to you.

clem844, 2020-03-19:
Yes, RPI_WIFI_ENABLE was set to 1. Setting it to 0 or removing it doesn't help. 
I already describe my setup here: https://github.com/agherzan/meta-raspberrypi/issues/581
But I realized it's probably more related to meta-updater than meta-rapberry, even if the error is similar.

patrickvacek, 2020-03-20:
I'm confused by one thing: your initial post in that thread says you are using a Raspberry Pi 1, but then in your local.conf I see `MACHINE = "raspberrypi3"`. Which version of the RPi are you actually using? I'm afraid if it's 1, we don't support it (see [here](https://docs.ota.here.com/ota-client/latest/supported-boards.html)).

I'm also curious how you ended up using the meta-blinkenlights layer. I would suggest removing that; it was a demo made in a couple days for fun, and it's not something that we maintain or intend to support.

Also, for CI and in regular usage, we typically only build `core-image-minimal`. I would expect `core-image-base` to work as well, but I'm curious if you can reproduce the problem with the more reduced target.

clem844, 2020-03-20:
The initial post in the thread wasn't from me. I'm using a raspberry pi 3. I was using `meta-blinkenlights` layer to make an LED blink to make the update visible for demonstration purposes. I don't think it interferes here, and I actually already removed it for these tests.
I'm trying to support TPM and thus need kernel modules. 
`core-image-minimal` builds without any problems.
Thanks for your support

patrickvacek, 2020-03-20:
> The initial post in the thread wasn't from me. I'm using a raspberry pi 3. I

Ah, sorry for missing that. Glad to hear that.

>  I'm trying to support TPM and thus need kernel modules.
> `core-image-minimal` builds without any problems.

Are you content with `core-image-minimal`, or do you need help with getting `core-image-base` to build correctly? Are the kernel modules and TPM support you need only in `core-image-base`, or can you add the packages you need on top of `core-image-minimal`?

clem844, 2020-03-20:
I'm not sure which exact modules are required here... All I know is that I already added some, but there isn't any /dev/tpm0 showing up...
For reference, I'm following this tutorial: [https://ubs_csse.gitlab.io/secu_os/tutorials/tpm_rpi.html](url)
They explicitly specify to use `core-image-base` instead of `core-image-minimal`

patrickvacek, 2020-03-23:
Are you still having problems building `core-image-base`? I just tried it with our zeus manifest from [updater-repo](https://github.com/advancedtelematic/updater-repo/) and I had no problem bitbaking for QEMU and Raspberry Pi 3. I'm not quite sure what the incompatibility is.

> I'm not sure which exact modules are required here... All I know is that I already added some, but there isn't any /dev/tpm0 showing up...

That's pretty hard for me to debug; I think you'd have to talk to your TPM manufacturer or the author of that guide for help on why the device doesn't show up. We can help with configuring aktualizr to use the TPM once it's there, though.

clem844, 2020-03-23:
Ok, after setting to zeus manifest and `repo sync` it works. However, my problem with the TPM comes from somewhere else.
I'm not sure if in the end I'll be using a TPM or another type of secure element. So no need further support for now. Thanks!

closed 2020-03-23 by clem844