advancedtelematic/meta-updater #706: kernel command line parameters are not being updated

jlarsen45, 2020-04-01:
I am using intel-corei7-64 with grub as bootloader.  If I change my kernel command line parameters the changes do not get updated when an update is performed.

Steps:
1. make image within Yocto project.
2. run image on board
3. view kernel parameters cat /proc/cmdline
4. change kernel command line parameters (OSTREE_KERNEL_ARGS)
5. make new image within Yocto project.
6. update board 
7. view kernel parameters cat /proc/cmdline

Expected result: command line parameters are the new parameters from new OSTREE_KERNEL_ARGS

Actual result:  command line parameters are stuck at initial values

patrickvacek, 2020-04-02:
Hello @jlarsen45! Unfortunately we don't have a good way of doing what you want to do right now. Currently, `OSTREE_KERNEL_ARGS` is only used in `classes/image_types_ota.bbclass`, which is the class used for building the local image, but not the image pushed to Treehub. The reason is that those kernel arguments are passed directly to `ostree admin deploy`, which for the local image we can call directly. But when updating an image via SOTA, the deploy is done via libostree and since the kernel arguments aren't stored in the image, we can't change them at that time.

If you have ideas on how to change that, I'd be interested to hear your thoughts.

Also, it's worth making sure we're on the same page, so a couple questions:
* What branch/version of meta-updater are you using?
* When you say "update board", do you mean over-the-air using Here OTA Connect?

And if you want to see exactly what is happening with that variable, I recommend looking at https://wiki.yoctoproject.org/wiki/Technical_FAQ#I_set_a_variable_but_it_doesn.27t_seem_to_be_having_an_effect.2C_how_do_I_fix_this.3F. But in this case, I think it just isn't getting used the way you want it to be.

jlarsen45, 2020-04-02:
Thanks @patrickvacek for your reply.

I am using meta-updater master branch (36c0d42d381)
I update my board through OTA connect "community edition"

I think I was too specific in my details.  In general, I want to be able to update kernel args with an over-the-air update. 
Does ostree have the ability to update kernel args with an update?

It seems like it should.  It already modifies the bootloader conf file (grub.cfg in my case).  Could it do an update to that file before it does the modifications that switch to boot the new rootfs?  Another idea is to keep a kernel arg file along with vmlinuz and initramfs.  There is karg handling code in libostree already correct?

patrickvacek, 2020-04-03:
> I think I was too specific in my details. In general, I want to be able to update kernel args with an over-the-air update.
> Does ostree have the ability to update kernel args with an update?

AFAIK ostree does support that, see for example https://github.com/ostreedev/ostree/issues/479#issuecomment-243735229. The problem you're going to face if trying to update with meta-updater is that our update client aktualizr does not currently support changing the kernel args when it deploys a new OSTree commit.

If you want to do it yourself, you might be able to try using the commandline run-mode options of aktualizr and then deploy the OSTree commit yourself, i.e.:

```
systemctl stop aktualizr
aktualizr check
aktualizr download
ostree admin deploy --karg=<...>
reboot
```

That *might* work.

patrickvacek, 2020-07-30:
Closing due to lack of feedback.

closed 2020-07-30 by patrickvacek