advancedtelematic/meta-updater #794: Gatesgarth branch

ricardosalveti, 2021-01-08:
I'm currently using meta-updater master (not latest due https://github.com/advancedtelematic/meta-updater/pull/791) just fine with latest OE gatesgarth, but it would be good to have a branch dedicated for it, as OE master moves forward with the next release.

After we land 791 (and possibly additional fixes related to it), we should be good to create a gatesgarth branch that would be initially in sync with master.

shr-project, 2021-02-04:
Agreed, now master also isn't compatible with latest oe-core/master, because after:
https://git.openembedded.org/openembedded-core/commit/?id=c56f3c9febc1732aa1302524c6c4da36f16bd1f7
the do_image_tar is failing, because of:
classes/image_types_ostree.bbclass:IMAGE_CMD_TAR = "tar --xattrs --xattrs-include=*"

```
| tar: --xattrs can be used only on POSIX archives
| Try 'tar --help' or 'tar --usage' for more information.
```

pattivacek, 2021-02-05:
Why was `gnu` chosen over `posix`? Oh well, whatever.

I'm not opposed to creating a gatesgarth branch, but we aren't using gatesgarth ourselves at this time, so we can't make guarantees about reliability at this time. I'd say until there is a breaking change made in master, it shouldn't really make a difference. We normally backport everything we can anyway.

shr-project, 2021-02-05:
@pattivacek I've asked RP in https://lists.openembedded.org/g/openembedded-core/message/147690

based on the answer I might fix it by submitting:
https://git.openembedded.org/openembedded-core-contrib/commit/?h=jansa/master&id=dafbf313bae7649ebad5098d6cad3a6218d3b5fa

pattivacek, 2021-02-05:
> I've asked RP in https://lists.openembedded.org/g/openembedded-core/message/147690
> 
> based on the answer I might fix it by submitting:
> https://git.openembedded.org/openembedded-core-contrib/commit/?h=jansa/master&id=dafbf313bae7649ebad5098d6cad3a6218d3b5fa

Oh great, thank you!

shr-project, 2021-02-06:
The tar format was changed from gnu to posix in master yesterday:
https://git.openembedded.org/openembedded-core/commit/?id=3ecea58f2a3382d9f4b410d6ad7089111334cb6f

pattivacek, 2021-02-08:
> The tar format was changed from gnu to posix in master yesterday:
> https://git.openembedded.org/openembedded-core/commit/?id=3ecea58f2a3382d9f4b410d6ad7089111334cb6f

Looks good, thank you!

shr-project, 2021-02-23:
There is another change for latest oe-core, but unfortunately not backwards compatible with gatesgarth (unless the LICENSE is changed as suggested), see:
https://github.com/advancedtelematic/meta-updater/pull/801

closed 2023-08-01 by ricardosalveti