advancedtelematic/meta-updater #817: Outdated OSTree branch reference in SRC_URI of ostree_git.bb files

stephaneberle9, 2021-08-06:
Affected branches (non-exhaustive): **rocko**, **warrior**

Current `SCR_URI` value in `recipes-sota\ostree\ostree_git.bb` file: 

```
SRC_URI = "gitsm://github.com/ostreedev/ostree.git;branch=master"
```

Actual branches in `https://github.com/ostreedev/ostree`: 

[image]

Change to fix the issue:

```
SRC_URI = "gitsm://github.com/ostreedev/ostree.git;branch=main"
```

pattivacek, 2021-08-07:
Thanks for the warning. IIRC rocko is not a supported branch anymore and AFAIK no one is using the warrior branch. I updated thud for my own needs; feel free to send PRs for the other two branches if you'd like.

By the way, do you have a script for detecting this sort of branch rename? :)

stephaneberle9, 2021-08-09:
>  feel free to send PRs for the other two branches if you'd like.

Done for `rocko`, `warrior`, `krogoth`, `sumo`, `pyro `and `morty`

> By the way, do you have a script for detecting this sort of branch rename? :)

No, I stumbled upon this issue because I've a rocko-based upstream project that was no longer compiling, and then just pocked around manually to find other branches with the same inconsistency ;)

pattivacek, 2021-08-09:
That's dedication! Thanks.

closed 2021-08-09 by pattivacek