advancedtelematic/meta-updater #828: Raspberry  pi3 not booting with meta-updater layer

nsmohan, 2022-05-09:
Hello. I have a raspberry pi3  Model B and I have created a minimal image using the yocto project. 

I initially created a image using the following layers: 

```
 BBLAYERS ?= " \
    /drv2/ostree/poky/meta \
    /drv2/ostree/poky/meta-poky \
   /drv2/ostree/poky/meta-yocto-bsp \
   /drv2/ostree/poky/meta-raspberrypi \
   /drv2/ostree/poky/meta-updater \
"
```

``` bitbake core-image-minimal ```

This generated a *.rpi-sdimg file which I then dd'd onto a SD card and was able to successfully run. 

I then updated my bblayers.conf file and added the following: 

```
BBLAYERS ?= " \
     /drv2/ostree/poky/meta \
     /drv2/ostree/poky/meta-poky \
    /drv2/ostree/poky/meta-yocto-bsp \
    /drv2/ostree/poky/meta-raspberrypi \
   /drv2/ostree/poky/meta-updater \
   /drv2/ostree/poky/meta-openembedded/meta-filesystems \
   /drv2/ostree/poky/meta-openembedded/meta-oe \
   /drv2/ostree/poky/meta-openembedded/meta-python \
"
```
I also updated my local.conf by adding the following: 

```
require conf/distro/sota.conf.inc
INHERIT += " sota"
```
I reran bitbake and this time flashed raspberrypi.wic file into the sd card via the dd tool. 

This time however when I try to boot I see the following error: 
``` Warning: no boot file name; using COA884C70.img```

This error basically shows up multiple times until a time out occurs. 

What am I missing ?

pattivacek, 2022-05-10:
You're missing https://github.com/advancedtelematic/meta-updater-raspberrypi. There was a long-term goal to merge the repos with dynamic layers, but we never got around to it.

Did you try following the [README](https://github.com/advancedtelematic/meta-updater#quickstart) and the [quickstart guide](https://docs.ota.here.com/getstarted/dev/raspberry-pi.html)?

BTW, [the Uptane namespace version of this repo](https://github.com/uptane/meta-updater) is more actively maintained, although you still have to use the meta-updater-raspberrypi repo from this namespace.

pattivacek, 2022-05-11:
> I used the flash-image.sh script to flash the image onto the SD card. Now I am seeing the device constantly rebooting right after uboot. Is there something I need to configure after the image has been built ?

That sounds like it isn't powered enough to boot. I've seen that problem lately with more and more boards, and I know for example the RPi4 basically requires wall power to boot successfully.

Otherwise, I'm afraid I don't know. I don't have a good way to test or debug myself anymore.